esp8266/Arduino #3779: Interrupts being blocked?

philbowles, 2017-10-31:
### Basic Infos

#### Hardware
Hardware:		Wemos D1 Mini
Core Version:      	2.3.0
IDE                          1.8.5

### Description

Tight loop seems to block interrupt

### Settings in IDE

Module:  			Wemos D1 Mini
Flash Size: 		4MB
CPU Frequency       80Mhz?

### Sketch
Notes:
1) pin 5 is looped back to pin 4 to create consistent pulse train into ISR
2) variable unguarded is deliberately NOT volatile
3) ISR continually increments variable unguarded @ 5kHz
4) loop retrieves value twice with 3 options for a short delay between them of approx 1ms
the "delta" between the two should be a count of how many times the ISR incremented during the delay
@ roughly 1msec delta should be approx 5.

with delayMicroSeconds(1176) the delta shows as 6 and program continues consistently

with delay(1) delta shows as 5 or 6 and pogram crashes with exception(0) time of crash is consistely between 15.8 and 16.8 seconds, irrespective of ISR frequency

with delay caused by call to wasteTime which lasts 1176us, delta=0 - no messages are ever shown (save for the heartbeat) hence it appears that the ISR is blocked during that time. How can this be? Is this "normal" behaviour? I thought the whole point if an interrupt was to -er...interrupt. If a yield is added in wasteTime, the delta (of course) reappears

I don't need "answers" to code correctness or alternative "solutions" as such - this is a deliberately contrived example to try to demonstrate a problem....

So:
1) under what circumstances do Interrupts normally block?
2) what is it that delay(1) does that causes a crash 15.8 sec on that delayMicroseconds does not? I presume its related to documented yield/no yield difference between the two, but what EXACTLY? can it be avoided?
3) why does the wastTime routine appear to get "optimised out" if the for/loop count is less than 1001? When that happens the calculted time is 0 usec

```cpp
#include<Ticker.h>

int unguarded=0;

Ticker heartbeat;

void ICACHE_FLASH_ATTR ISR1(){
  unguarded++;
}

void wasteTime(){
  for(float i=0.0;i<1001.0;i++); // as close to 1000us = 1ms as I can get
}

void setup(){
  Serial.begin(74880);
  Serial.printf("T=%d Setup\n",millis());
  // confirm wasteTime value:
  int T=micros();
  wasteTime();
  Serial.printf("wasted %d microseconds\n",micros()-T);
  // Pin 4 is looped back from pin 5
  pinMode(5,OUTPUT);
  analogWriteFreq(5000);
  analogWrite(5,512);  pinMode(4,INPUT);
  attachInterrupt(4,ISR1,RISING);
  // just a confirmation that unguarded is being continually incremented by INTs
  heartbeat.attach_ms(1000,[](){
    Serial.printf("T=%d HB ug=%d\n",millis(),unguarded);
    });
}

void loop(){
  int temp1=unguarded;
//  delayMicroseconds(1176);
//  delay(1);
  wasteTime(); // BLOCKS INTERRUPTS!!!
  int temp2=unguarded;
  if(temp1!=temp2) Serial.printf("T=%d  delta=%d\n",millis(),temp2-temp1);
}
```

### Debug Messages

```
T=15850  delta=5
T=15853  delta=5
T=15855  delta=5

Exception (0):
epc1=0x40202b94 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffc30 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffdd0:  40106c70 3ffee30c 000035af 00000000  
3ffffde0:  ffffffff 00000020 3ffee30c 3fffc230  
3ffffdf0:  fffffc00 3fffc200 00000020 00000022  
3ffffe00:  3fffc200 40106c38 3fffc258 4000050c  
3ffffe10:  40004378 00000030 00000016 ffffffff  
3ffffe20:  60000200 00000008 ffffffff 80000000  
3ffffe30:  20000000 3fff0560 80000000 203fc1e0  
3ffffe40:  80000000 3fffc6fc 00000001 3fff0564  
3ffffe50:  00000114 003fc1e0 60000600 00000030  
3ffffe60:  4021cd6b fc000000 00000020 4010702c  
3ffffe70:  40106cd4 40106c38 3fffc230 4000050c  
3ffffe80:  ffffffff 00000020 00000017 ffffffff  
3ffffe90:  00000000 00000000 0000001f 40101361  
3ffffea0:  4000050c 40106c38 3fffc230 4000050c  
3ffffeb0:  40000f68 00000030 00000017 ffffffff  
3ffffec0:  40000f58 00000000 00000020 00000000  
3ffffed0:  feefeffe 3ffee30c 000035af 00000000  
3ffffee0:  3ffee30c 3ffe9184 3ffee30c 3fffdab0  
3ffffef0:  00000000 3fffdad0 3ffee47c 00000030  
3fffff00:  00000000 400042db 00000064 60000600  
3fffff10:  40004b31 3fff0384 000002f4 003fc000  
3fffff20:  40102292 3ffe8930 3ffe90d0 40107230  
3fffff30:  4021aa09 3ffe90d0 3ffe8930 00f1f3ba  
3fffff40:  3fff0384 00001000 4021aea2 00000008  
3fffff50:  00000000 00000000 4021af4f 3ffe9184  
3fffff60:  3ffe8930 0032afec 3ffe8930 40201018  
3fffff70:  402034c1 3ffe9184 3ffe8930 00f1eeca  
3fffff80:  40203506 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffe8958 3fffdad0 3ffee47c 40202527  
3fffffa0:  40000f49 0000b829 3fffdab0 40000f49  
<<<stack<<<

Exception 0: Illegal instruction
Decoding 29 results
0x40202b94: ISR1() at C:\Users\phil\Downloads\esp8266_mqtt/esp8266_mqtt.ino line 8
0x40106c70: interrupt_handler at C:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_wiring_digital.c line 119
0x40106c38: interrupt_handler at C:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_wiring_digital.c line 113
0x4021cd6b: pp_enable_noise_timer at ?? line ?
0x4010702c: timer1_isr_handler at C:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_timer.c line 40
0x40106cd4: interrupt_handler at C:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_wiring_digital.c line 135
0x40106c38: interrupt_handler at C:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_wiring_digital.c line 113
0x40101361: ets_timer_disarm at ?? line ?
0x40106c38: interrupt_handler at C:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_wiring_digital.c line 113
0x40102292: spi_flash_read at ?? line ?
0x40107230: pvPortZalloc at C:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/heap.c line 33
0x4021aa09: pm_set_sleep_time at ?? line ?
0x4021aea2: pm_get_sleep_type at ?? line ?
0x4021af4f: pm_get_sleep_type at ?? line ?
0x40201018: delay_end at C:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_wiring.c line 40
0x402034c1: ets_timer_handler_isr at ?? line ?
0x40203506: ets_timer_handler_isr at ?? line ?
0x40202527: loop_task at C:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56

```

devyte, 2017-10-31:
@your I see the following:
- your ISR1 function needs to have ICACHE_RAM_ATTR  instead of ICACHE_FLASH_ATTR. The random crash is likely caused by that.
- if you need accurate timing, you probably should use delayMicroseconds instead of wasteTime()
- delay() yields the context of the loop to allow servicing of the wifi stack, while delayMicroseconds() does not yield and just keeps the cpu doing nothing, similar as your intent with wasteTime(). The former is more efficient because of the wifi servicing, the latter is more accurate.
- wasteTime() is probably getting optimized out by the compiler. I have no idea what happens in that case when the interrupt happens, but based on your report, I'm going to guess that in this case only the body function gets dropped, and the function ends up as empty. When you add a call to yield(), you are in effect adding a side-effect, and then the compiler can't drop it anymore. If you want to avoid that, it probably needs the i variable to be global and declared as volatile float i or something along those lines (I'm guessing based on some past experience, I'm not a compiler expert).
- why exactly is variable unguarded deliberately NOT volatile?

Overall, I don't see a problem here. Closing.

closed 2017-10-31 by devyte

philbowles, 2017-11-01:
Devyte,

Thanks for the reply, but I maybe I didn't make my point well. When wasteTime isn't optimised out, it blocks the interrupts. This behaviour is the same irrespective any ICACHE attributes on the ISR or whether the int is volatile. 

I know about the accurate timing, I wrote wasteTime to deliberately NOT use any delay functions. I don't want accurate timing, or  "sensible" code, I am deliberately trying to create a code example that demonstrates a problem.

I understand the difference between delay and delayMicroseconds and I understand why adding yield changes the behaviour. 

What I don't understand, and what is the whole point of the problem and what you haven't answered is why interrupts are blocked during wasteTime.

I know wasteTime is a "silly" piece of code, I'd never use it in real life. I wrote it to show how Interrupts can be blocked but i dont known HOW or WHY it is doing it - that's why I'm trying to raise.

igrr, 2017-11-01:
I think that the interrupts are not actually being blocked. Just the compiler sees that calling `wasteTime` function will not change the value of `unguarded`, so it optimizes away the second assignment (`int temp2 = unguarded;`). If you want the second assignment to take place, the `unguarded` variable must be declared as `volatile`.

philbowles, 2017-11-01:
@igrr - thank you - that was exactly what I was looking for - its why I
deliberately omitted it to show the result / problems of not having it.
It's why I called it "unguarded", i.e. not protected from simultaneous
access...I just didn't have the explanation for the mechanism. Now I do!

I'm just trying to come up with an example for a blog post of problems with
"unguarded" variable in this type of situation and It's not going to read
well if the conclusion is: "so I know this doesn't work, but I can't
explain why".

Now I can write a sensible (and much more informed!) conclusion. thanks
again.

PS please be assured this is NOT normally how I write code! I usually write
it to work, not break! :)

Phil

philbowles, 2017-11-01:
What I don't still fully understand is why adding a yield to "wasteTime" or replacing it with either delay or delayMicroseconds prevents the optimisation. Is it because the compiler "thinks" that any of those things _could_ affect the non-volatile variable?

igrr, 2017-11-01:
Yes, because `delay`, `delayMicroseconds` are defined in a different translation unit, and `unguarded` is a common variable (i.e. not explicitly declared as extern or static) — so depending on what happens at link time it may be modified by delay et al.
You may see optimization behaviour change if you declare `unguarded` as static.

philbowles, 2017-11-01:
@igrr. Once again, an excellent response. Thank you.