esp8266/Arduino #3781: Hi, I am struggling with ESP8266 for past 2 weeks.  I have made all the connections as you and several other tutorials have mentioned. I am using an external power source for the ESP.  I am not getting any response for AT. The serial monitor just remains blank. I have tried with different baud rates(9600 & 115200). I have noticed that whenever I connect CH_PD pin to 3.3V, my onboard red LED of ESP goes off. When I remove the CH_PD pin from 3.3V, the LED turns back on. All the tutorials have mentioned that the CH_PD pin must be kept at 3.3V. But on making this connection my LED goes off and the voltage between GND&VCC and also GND&CH_PD drops to about 0.8V when measured with a multimeter. Can you please suggest what needs to be done to get the ESP8266 working

mihirdharap, 2017-11-01:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

rick-bp, 2017-11-01:
Hi! REST is pluged in?

Check this: [http://buger.dread.cz/images/esp12f-basic-connections.jpg](http://buger.dread.cz/images/esp12f-basic-connections.jpg)

rick-bp, 2017-11-01:
To power on ESP12, you need:
REST and CH_PD must conected to 3.3V
GPIO15 must pull down to GND with 10k
GPIO0 must pull down to GND when you will send flash.

mihirdharap, 2017-11-01:
I am working with esp 01 wich is having just 8 pins. I have tried with
both, RST connected to 3.3v and rst left open. In both cases same problem
persists. As soon as i connect ch_pd to 3.3v, onboard red led turns off.

rick-bp, 2017-11-01:
Check this: [http://simba-os.readthedocs.io/en/latest/_images/esp01-pinout.png](http://simba-os.readthedocs.io/en/latest/_images/esp01-pinout.png)

May be, your ESP is burn out.

mihirdharap, 2017-11-02:
> pinout.png

Hi,
To check the possibility of esp burning out, i bought a new esp module
today, but exactly same problem persists. I am not understanding what am I
doing wrong.

mihirdharap, 2017-11-02:
Hi,
To check the possibility of esp8266 burning out, i bought a new esp module
today. However, exact same problem persists. Following are the connections
that I have done:

ESP GND -> external GND
ESP Vcc -> external 3.3V
ESP RST -> external 3.3V
ESP CH_PD -> external 3.3V

My external source is a 5V power source with 2A current rating. I have used
three 1k resistors to get 3.3V. I had also checked the voltages with
multimeter before connecting to the esp module. All tge voltages were
between 3.27V to 3.3V. I am really stuck at this and I'm not understanding
what am I doing wrong. Please any help would be greatly appreciated!

Pablo2048, 2017-11-02:
So You are trying to tell us, that Your power supply is NOT 3.3V, but 5V with 1k resistor in series?

mihirdharap, 2017-11-02:
Basically, a voltage divider. As shown in the figure given in the link below.

http://www.akafugu.jp/images/schematics/divider.png

Pablo2048, 2017-11-02:
Hmm, but this is definitely NOT proper way to power such kind of IC... Do get right power supply first.

devyte, 2017-11-02:
Not the right place to request help for wiring.
Closing per #3655 and due to horrible title.

devyte, 2017-11-02:
And no, a voltage divider is NOT the  right way to provide a power source for any IC.

closed 2017-11-02 by devyte

philippe-gregoire, 2017-11-02:
Just to explain a bit, a voltage divider works based on ohm's law, i.e. the middle point of your resistors will present a voltage that depends on the R1/(R1+R2) ratio (here you have R1=2k, R2=1k) . When you connect your ESP to that middle point, it comes in // of R1, so you basically change the value of R1, to something much lower, hence the voltage drop you see.
Voltage dividers are ok to get reference voltage when there is no load (High impedance), not when there is current drain.
You should use a regulator for this (e.g. AMS1117).

mihirdharap, 2017-11-02:
Thanks for the explanation!

mihirdharap, 2017-11-03:
It worked with a voltage regulator. Thanks a lot!!

Utkarsha243, 2019-04-01:
Hi
I am also working with ESP8266-01 module. I have connected ESP8266 with arduino. I want to measure current drawn by ESP8266. But I don't know how to measure it because other components like LEDS, resistors all are connected to breadboard directly and measuring their current by multimeter is easy. But ESP8266 is connected to arduino, then how to measure its current.