esp8266/Arduino #3782: Incompatible function definitions between <=SDK 2.0 and >=SDK 2.1

slaff, 2017-11-01:
In SDK 2.1 and later two functions coming from the BootROM have changed definition. 

# The first one is:
void	os_delay_us(uint16	μs)

which was before that defined as:
void	os_delay_us(uint32	μs)

The actual code of the function did not change. `os_delay_us` is leading to `ets_delay_us` which is located at `40002ecc` which according to the [memory map](https://github.com/esp8266/esp8266-wiki/wiki/Memory-Map) is the BootROM area. Since the code did not change does it mean that the definition before 2.1 was wrong or in SDK 2.1 and later it is wrong?

# The second one is `ets_isr_attach` 
In SDK 2.1 and later it is defined as in include/ets_sys.h line 67
`void ets_isr_attach(int i, ets_isr_t func, void *arg);` 

where `ets_isr_t`

`typedef void (* ets_isr_t)(void *);`

Using  `xtensa-lx106-elf-readelf -s file.out` I can see that this function is defined at address `40000f88` which is again in BootROM. The SDK 2.1+ definition is conflicting with the [ESP-GDBStub](https://github.com/espressif/esp-gdbstub/issues/17) definition. So my question is where is the error being introduced?

@igrr Can you shed some light on this?

igrr, 2017-11-02:
I don't know where the definition in esp-gdbstub came from. In theory, interrupt handler in xtos is declared as a function with variable number of arguments. But i never saw more than one argument being passed to it.

Regarding os_delay_us, new definition is the correct one. The function indeed takes uint32_t as an argument.

slaff, 2017-11-02:
> Regarding os_delay_us, new definition is the correct one. The function indeed takes uint32_t as an argument.

@igrr Ok, if I understood you correctly this means that the current definition in the SDK is just wrong 
https://github.com/espressif/ESP8266_NONOS_SDK/blob/master/include/osapi.h#L33 because it takes uint16_t as an argument. Can you confirm this so that I can open an issue in the `ESP8266_NONOS_SDK` project?

igrr, 2017-11-03:
See https://github.com/espressif/ESP8266_NONOS_SDK/issues/31

earlephilhower, 2019-07-25:
This is/was an issue in the NONOS-SDK and needs to be taken up there.  Here in the Arduino core we can only take what we're given from there...  Closing, but I do suggest following up in https://github.com/espressif/ESP8266_NONOS_SDK/issues/ if there's still some issue.

closed 2019-07-25 by earlephilhower