esp8266/Arduino #3785: Watchdog didn't work or i cannot use it?

martinius96, 2017-11-01:
#ISSUE
@igrr
Hello there, i tried these ESP. functions for watchdog but i dont know how it works.. I put in setup that:
`ESP.wdtEnable(WDTO_`2S);
At the end of loop i take that:
`ESP.wdtFeed();`
My loop working time is circa 6 seconds for one loop. I enabled wdt for 2 seconds and it didnt reset my NodeMCU v3 Lolin. I only want to reset my board to see if it is working if code is running more than 2 seconds.. Where is problem? Maybe i bad think about how it functions I want that: if my board lags (stops working) it will reset my board automatically. I am using HTTPS requests to webpage (IoT house) for upload and download values and full automatization. It takes 6 seconds together. Thanks for help!

WereCatf, 2017-11-02:
> My loop working time is circa 6 seconds for one loop. I enabled wdt for 2 seconds and it didnt reset my NodeMCU v3 Lolin.

That's not how the watchdog works. The watchdog is fed every time loop() is run, every time you call delay() and many libraries also include watchdog-feeds in long-running functions.

martinius96, 2017-11-02:
This is my code. How can i integrate watchdog to it? Can you help me?
`#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>
#include <OneWire.h>     
#include <Adafruit_Sensor.h>
#include <DHT.h>
#include <DHT_U.h>
#include <DallasTemperature.h>       //KNIZNICA PRE TEPLOTNE CIDLA DS18B20
#define ONE_WIRE_BUS 14             //D5 DEFINICIA PINU AKO ZBERNICE PRE ONEWIRE ZARIADENIA.. TU ZBIERAME UDAJE
OneWire oneWire(ONE_WIRE_BUS);       //ONEWIRE ČÍTAŤ IBA NA PORTE DEFINOVANOM VYSSIE
DallasTemperature sensors(&oneWire); //PRIRADENIE SENZOROV DALLAS DS18B20 NA ONEWIRE ZBERNICU
#include <SPI.h>
DHT_Unified dht(5, DHT22); //DHT22 na PINE D1
int pirState = LOW;
int valpir = 0;
const int zavlaha = 12; //D6 vystup
const int termostat = 13; //D7 vystup
const int pir = 4; //D2 vstup
const char* ssid = "moj-sinet-2929";
const char* password = "chefrolet";
const char* host = "mywebserver.com"; //bez https a www
const int httpsPort = 443;
const char* fingerprint = "13 9f 87 1d b1 85 be e6 bd 73 c1 8d 04 63 58 99 f0 32 43 92";
void setup() {
 
  sensors.begin();
dht.begin();
  pinMode(pir, INPUT);
  pinMode(zavlaha, OUTPUT);     // declare sensor as input
  pinMode(termostat, OUTPUT);     // declare sensor as input

  Serial.begin(115200);
  Serial.println();
  Serial.println("pripajam na ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }  Serial.println("");
  Serial.println("WiFi pripojene");
  Serial.println("IP adresa: ");
  Serial.println(WiFi.localIP());
   
 
}

void send_pir() {
  valpir = digitalRead(pir);
  if (valpir == HIGH) {
    if (pirState == LOW) {
      WiFiClientSecure client;
      if (!client.connect(host, httpsPort)) {
        return;
      }
      if (client.verify(fingerprint, host)) {
      }
      else {
      }
      String url = "/celi/system/nodemcu/zapispir.php?pir=Pohyb";
      client.print(String("GET ") + url + " HTTP/1.1\r\n" + "Host: " + host + "\r\n" + "User-Agent: NodeMCU\r\n" + "Connection: close\r\n\r\n");
      Serial.println("Odoslane: Pir pohyb");

      pirState = HIGH;
    }
  } else {

    if (pirState == HIGH) {
      WiFiClientSecure client;
      if (!client.connect(host, httpsPort)) {
[... 139 lines cut ...]
  }
  String line = client.readStringUntil('\n');
  if (line == "ZAP") {
    Serial.println("Zapni termostat");
    digitalWrite(termostat, LOW);
  } else if (line == "VYP") {
    Serial.println("Vypni termostat");
    digitalWrite(termostat, HIGH);
  }
}
//-----------------------------------------------------------------------------------

void loop() {
  send_teplota();
  send_zavlaha();
  send_pir();
  calibrate_zavlaha();
  calibrate_termostat();

  get_zavlaha();
  get_termostat();
 

}
`

devyte, 2017-11-02:
@

closed 2017-11-02 by devyte