esp8266/Arduino #3788: Активация функции FlowControl

mcmega, 2017-11-03:
Спасибо за библиотеку и Ваш труд!
Подскажите, планируется ли добавление нормальной библиотеки в этот проект по работе с UART?
Дело в том, что ESP поддерживает контроль потока и имеет прерывания по работе с UART на аппаратном уровне. Я сделал WiFi программатор микроконтроллеров AVR. Но там используется контроль потока. Если искуственно создавать задержки, то все работает, но теряется скорость , хотелось бы использовать аппаратные возможности ESP, а не универсальную библиотеку Arduino.

latonita, 2017-11-03:
Во-первых, тут принято писать на английском.
At second, i think esp is lacking some of interrupts which are widely used in other MCUs. so in any way there shall be some software magic happening.

devyte, 2017-11-03:
@mcmega There are developers from many different countries here, but the common language is English. Please restrict comments and issues to that.
There is currently no plan to add hardware flow control to HardwareSerial or SoftwareSerial. That doesn't mean that it can't be done.
I invite you to investigate and implement this yourself, and propose a PR for review. The high level part is in Hardware.cpp|h, the low level code is in uart.c|h.
I could leave this request open, but I suspect there would be _very_ few users of this feature. Therefore, if you do not intend to pursue this yourself, please close this issue.

closed 2017-11-03 by mcmega