esp8266/Arduino #3789: Hardware Flow - RTS is always high

latonita, 2017-11-03:
#### Hardware
Hardware:			ESP-12f, NodeMCU black with ESP-12f
Core Version:      	2.3 stable and 2.4 rc2

### Description

Hi, i would like to use RTS signal as "TX enable" for rs485 half-duplex communication.
I turned on RTS in esp with following function:
`PIN_FUNC_SELECT(PERIPHS_IO_MUX_MTDO_U, FUNC_U0RTS);`
and i expected that RTS pin (GPIO15) will be high when HardwareSerial is transmitting.

However what i observe is that RTS pin is just always high. No matter what i do with serial.

So what i have to do is to manually work with GPIO15. make it high before transmission and make it low just after transmission. which is somewhat okay, but wrong.

### Sketch

```cpp

 #define TXE 15 // TX ENABLE PIN for rs485 transceiver
void startTx() {digitalWrite(TXE, 1);}
void stopTx()  {digitalWrite(TXE, 0);}

void setup() {
    Serial.begin(9600);
    // option 1. not working. tried to do it before Serial.begin as well.
    //PIN_FUNC_SELECT(PERIPHS_IO_MUX_MTDO_U, FUNC_U0RTS);
    // option 2. manual workaround
    pinMode(TXE, OUTPUT);
    stopTx();
}

void loop() {
    static int i = 0;
    Serial.print('.');
    Serial.print(i++ % 10);

    startTx();
    for (char c = 32; c < 127; c++) {
        Serial.write(c);
    }
    Serial.println();
    Serial.flush();
    stopTx();
    delay(1000);
}
```

devyte, 2017-11-03:
@latonita Is this a bug report that the register setup doesn't work, or a feature request for hardware flow control? If the first, this is not the right place to report it, it would be better to report it to Espressif.

From a quick google, I suspect that the register setup that you are doing is somehow wrong. That's just a guess, I know next to nothing about the low level internals of the ESP.

If this is a feature request, as I explained in #3788 , there is currently no plan to implement hardware flow control, but that doesn't mean it can't be done.
I invite you to investigate and implement this yourself, and propose a PR for review. The high level part is in Hardware.cpp|h, the low level code is in uart.c|h.
Some links from a quick google that may help as a starting point: 
- [old AT firmware attempt](https://github.com/igrr/atproto/issues/8)
- [topic at Espressif](http://bbs.espressif.com/viewtopic.php?f=7&t=27&sid=2507a4bdd37b058bdc2680a943499b50)
- [ESP8266 uart registers](https://github.com/esp8266/Arduino/blob/master/cores/esp8266/esp8266_peri.h#L189)

You are only the 2nd person I have seen asking about hardware flow control. I suspect there would be  _very_ few users of this. If you intend to pursue this, I suggest a collaboration with the author of #3788 . If not, please close the issue.

closed 2017-12-27 by igrr

latonita, 2017-12-28:
Igrr, any comments? Thanks

ondra-simons, 2018-04-26:
Good day all.
I have an existing project that uses a PIC micro controller connected to ESP8266 implementing flow control using the espressif AT shield.  I have started some work towards building/using the Auto connect web configuration tool. Now that I have it working along with a web client I would like to implement flow control so as to keep with the existing design. I have spent a few days searching for an existing design/implementation.  Can someone help me. I would much appreciate it. Thanks in advance