esp8266/Arduino #3791: Handling POST data from AJAX  2.4.0-rc2

kapyaar, 2017-11-03:
### Basic Infos

#### Hardware
Hardware:		ESP-12
Core Version:      	2.4.0-rc2

### Description

I have a form that inputs SSID and Password. To handle special characters in ssid and or passwords (especially "&"), I would encode the String before submitting it. In 2.3.0, it was working ok. But Trying to upgrade to 2.4.0-rc2, I have a problem.

In my ajax, I did not have any "Content-Type" Set. This leads to the ESP spitting the POST Data as "Plain", and unparsed.

So I tried 

```xmlhttp.setRequestHeader("Content-Type", "application/x-www-form-urlencoded");```

Now, the data gets parsed, but then comes trouble. Say my post data looks like this.

```ssid=his&herWiFi&password=her&hisPassword```

I encode just the SSID and Password before sending, so it looks like  
```ssid=his%26herWiFi&password=her%26hisPassword```

But now on the ESP, server.arg("ssid") gives "his", and password gives "her" as it cuts off after the "&"

I also tried
```xmlhttp.setRequestHeader("Content-Type", "multipart/form-data");```

But that leads to some boundary error messages during  parsing. 

Any Idea on how to handle this situation?

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		4MB/1MB
CPU Frequency:	80Mhz
Flash Mode:		qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch

```cpp

for ( uint8_t i = 0; i < server.args(); i++ ) {
        if (server.argName(i) == "ssid") {
          DEBUG_PRINT("Got RAW SSID ");
          DEBUG_PRINT(server.arg("ssid"));
          //ssid = urldecode(server.arg("ssid"));//Does not seem to need decoding.
          ssid = server.arg("ssid").c_str();;
        }
        if (server.argName(i) == "password") {
          DEBUG_PRINT(" and pass: ");
          DEBUG_PRINT(" and "); DEBUG_PRINTLN(server.arg(i));
          //pass     = urldecode(server.arg(i)).c_str();;//Does not seem to need decoding.
          pass     = server.arg(i).c_str();
        }
      }
```

### Debug Messages

```
messages here
```

devyte, 2017-11-12:
@kapyaar does PR #3313 fix this?

AbrahamEsensors, 2017-11-12:
@devyte YEs, that looks like it. Parsing first, and then decoding should solve this! If I update the cores using staging, is this update available now?

kapyaar, 2017-11-18:
@devyte YEs, that looks like it. Parsing first, and then decoding should solve this! If I update the cores using staging, is this update available now?

devyte, 2017-12-30:
Closing via #3313 .

closed 2017-12-30 by devyte