esp8266/Arduino #3799: persistent inconsitancies and checks

tablatronix, 2017-11-05:
All functions that use persistent and config equal or saves needs to have default/current conditional based on persistent. There are a few, I will make note of them.

`ESP8266WiFiAPClass::softAP`
`ESP8266WiFiSTAClass::begin`

This also affects functions that check mode, but save persistent
```C++
ESP8266WiFiGenericClass::mode

    if(wifi_get_opmode() == (uint8) m) {
        return true;
    }
```

It is also special note, that the docs say
This API can be called only if ESP8266 Station is enabled.
I am not sure if this applies to *_default config (docs only mention flash) or what the consequences are, it may be necessary to add a sanity check if it is prohibited.

The main problem with persistent is that is assumes current config has not been changed from default, which it can be by toggling persistent on and off. So both need to be checked in comparisons, this has been a major thorn in some projects that preserve default config but switch in and out of alternative configs , for example captive portals, wifimanager etc.

tablatronix, 2017-11-05:
This is what I am thinking

https://github.com/esp8266/Arduino/pull/3798

tablatronix, 2017-11-05:
I can provide test case or scenarios, but essentially allow persistent to be disabled and wifi modes and config to be changed at will. Some of these checks do not allow this because they check only current configs and abort.

devyte, 2018-03-12:
Closed via #3798 .

closed 2018-03-12 by devyte