esp8266/Arduino #3800: vcc = ESP.getVcc() inaccurate in 2.4.0-rc2 version?

sam77277, 2017-11-05:
### Basic Infos

#### Hardware
Hardware:			ESP-12
Core Version:      	2.4.0-rc2

### Description

I connected A0 pin to 3V pin and connected power to 3V pin. The value of "vcc = ESP.getVcc()" in 2.4.0-rc2 version is inaccurate, but the value in 2.3.0 version was accurate that was the same voltage as power. It was wired.

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?SERIAL?
Reset Method:		?nodemcu?

### Sketch

```
ADC_MODE(ADC_VCC);

void setup() {
   Serial.begin(115200);
}

void loop() {
int vcc = ESP.getVcc();//readvdd33();
Serial.println(vcc);
delay(1000);
}

```

igrr, 2017-11-05:
IIRC you have to keep A0 disconnected if you wish to use getVdd.

sam77277, 2017-11-05:
Despite whether I connected or disconnected A0, the Vdd was 10% difference  between using 2.4.0-rc2 or 2.3.0 version. If I want to get the accurate reading, I have to use 2.3.0 and connect A0 to 3V.
I still confused.

comino, 2017-11-06:
`int vcc = ESP.getVcc();`

According to the nonOS datasheet : 
`Power voltage of VDD33; unit: 1/1024 V`

So to get the 'correct' voltage in V you need to divided be 1024. Like: 
`float vccVolt = ((float)ESP.getVcc())/1024;`

Hope that helps.

tablatronix, 2017-11-21:
The wemos d1 mini has a resistor divider on A0, so your test is flawed.

devyte, 2017-11-21:
Previous comment is correct. Closing due to user error.

closed 2017-11-21 by devyte

ishwariyjoshi, 2020-01-18:
can we use ESP.getVcc();  funtion in ESP32 to read VCC voltage ?