esp8266/Arduino #3801: ESP8266WiFi, sending HEX codes

Rino664, 2017-11-05:
Hi All, I discover a strange behavior on ESP8266 with this library.
Watching via WireShark and sending 23 HEX bytes, the first HEX code is send alone in one packet and the rest 22 in another packet.
I use this HEX code to send a command to a PLC; unfortunately seems that don't like this way to send.
With the PLC software infact the whole HEXs are sent in one shot.

I use:
#### Hardware
Hardware:	ESP8266-12E
Core Version:   2.4.0-rc2

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP-12E Module)
Flash Size: 		4MB
CPU Frequency:	160Mhz?
Flash Mode:		AVRISP MKII
Flash Frequency:	??
Upload Using:		SERIAL
Reset Method:		??

This is the Code I use:

#include <ESP8266WiFi.h>

const char* ssid  = "MyAccessPoint";
const char* password = "MyPassword2AP";
const char* host = "192.168.1.116";  //PLC IP address
const int httpPort = 1280;     // Port to communicate with PLC

void setup() {
  Serial.begin(115200);
  delay(10);
  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  Serial.print("connecting to ");
  Serial.println(host);
  
  WiFiClient client;
  
  if (!client.connect(host, httpPort)) {
    Serial.println("connection failed");
    return;
  }
  
 // this is the string od HEX codes to send -> 50 00 00 FF FF 03 00 0E 00 10 00 01 14 00 00 00 00 00 A8 01 00 21 00
 byte manda[]={0x50,0x00,0x00,0xff,0xff,0x03,0x00,0x0E,0x00,0x10,0x00,0x01,0x14,0x00,0x00,0x00,0x00,0x00,0xA8,0x01,0x00,0x21,0x00}; // This array contains all HEX code to send
 
      Serial.println("Send data");

      for (int i =0; i<sizeof(manda);i++){
        client.write (manda[i]);
       // Serial.print(manda[i],HEX);
          }
      
      Serial.println("Data has been sent");
      client.stop();
  }

void loop() {
}

I don't know why the library divide in two packets the sending; if you know some other tricks, please let me know

Thanks
Rino.

Pablo2048, 2017-11-05:
IMHO this behavior is because of dealyed ACK algorithm, used in Windows OS. Windows OS send ACK for TCP datagram every second packet or after 200ms idle period.

Rino664, 2017-11-05:
Many thanks Pablo, any suggestion on how to write @ once the whole HEX codes ?
I tried to use //client.write (manda);
But I got an error....

Thanks
Rino.

Pablo2048, 2017-11-05:
Probably the best answer to this kind of "how to ..." is reading the documentation and header/source code of library (in this case https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/client-class.html).
client.write(manda, sizeof(manda));

Rino664, 2017-11-05:
Pablo, you got it in one shot !!
In the way you suggest I got all in once !
500000ffff03000e00100001140000000000a801002100

Many thanks for your support, tomorrow I will try with PLC...
Rino.

Pablo2048, 2017-11-05:
@Rino664 glad to hear ;-) so please close this issue...

closed 2017-11-05 by Rino664