esp8266/Arduino #3802: Can't connect to thingspeak with WiFiClientSecure.h

Staubgeborener, 2017-11-05:
### Basic Infos

#### Hardware
Hardware:			ESP8266
Core Version:      	?2.1.0-rc2?

### Description

I try to upload some data to thingspeak (thingtweet), but i can't connect to thingspeak. It works fine with ``WiFiClient.h``, but not with ``WiFiClientSecure.h``. Is ``WiFiClientSecure.h`` not working with thingspeak? So basically, ``client.connect("184.106.153.149", 80)`` does not work. The connection between ESP8266 and Router is working well.

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP-12E Module)
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			?qio?
Flash Frequency:	 115200 baud
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

// https://github.com/esp8266/Arduino/blob/master/doc/esp8266wifi/client-secure-examples.rst

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>

WiFiClientSecure client;

const char* ssid = "ssid";
const char* password = "passwd";
const char* wiki_host = "en.wikipedia.org";
const int httpsPort = 443;

// Use web browser to view and copy
// SHA1 fingerprint of the certificate
const char* fingerprint = "58 66 84 EF 77 3E A0 B8 5F 23 38 73 CB 46 10 E8 D0 E0 8C B3";

String API = "api";

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.print("connecting to ");
  Serial.println(ssid);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  
}

void loop() {
  
  // Use WiFiClientSecure class to create TLS connection
  //Serial.print("connecting to ");
  //Serial.println(host);
  if (!client.connect(wiki_host, httpsPort)) {
    //Serial.println("connection failed");
    return;
  }

  if (client.verify(fingerprint, wiki_host)) {
    Serial.println("certificate matches");
  } else {
    Serial.println("certificate doesn't match");
  }

  //using xml, 'cause json can't handle special characters and all that stuff
  String url = "/w/api.php?&action=query&format=xml&prop=extracts&generator=search&exsentences=1&exintro=1&generator=random&exlimit=10&explaintext&grnnamespace=0";
  //Serial.print("requesting URL: ");
  //Serial.println(url);

  client.print(String("GET ") + url + " HTTP/1.1\r\n" +
               "Host: " + wiki_host + "\r\n" +
               "User-Agent: BuildFailureDetectorESP8266\r\n" +
               "Connection: close\r\n\r\n");

 // Serial.println("request sent");
  while (client.connected()) {
    String line = client.readStringUntil('\n');
    if (line == "\r") {
      //Serial.println("headers received");
      break;
    }
  }
  String line = client.readStringUntil('\n');

    String tweet = line.substring(line.indexOf("preserve"));
    tweet.remove(0, 10);
    int lastIndex = tweet.length() - 39;
    tweet.remove(lastIndex);
    tweet.replace("&quot;", "'");    
    if((tweet.length() <= 140) && (tweet.length() > 30) && (tweet.indexOf("refer to") == -1)){
        Serial.println(tweet);
        if (client.connect("184.106.153.149", 80))
        {
          Serial.println("do this");
          client.print("GET /apps/thingtweet/1/statuses/update?key=" + API + "&status=" + tweet + " HTTP/1.1\r\n");
          client.print("Host: api.thingspeak.com\r\n"); 
          client.print("Accept: */*\r\n"); 
          client.print("User-Agent: Mozilla/4.0 (compatible; esp8266 Lua; Windows NT 5.1)\r\n");
          client.print("\r\n");
        }

        delay(60000);
    }
    
  
}

```

### Debug Messages

```
messages here
```

vdeconinck, 2017-11-05:
Err, maybe I don't understand, but if you try to connect securely, shouldn't it be client.connect("184.106.153.149", **httpsPort**);
?

Staubgeborener, 2017-11-06:
@vdeconinck No. I have to use ``WiFiClientSecure.h`` for a secure connection to wikipedia (they use https -> port 443). When i try do get a 'unsecure' connection to wikipedia (port 80) it won't work. And thingspeak seems to NEED port 80 for a connection. I thought to myself 'Ok, in this case i just make a connection to wikipedia with 443 and to thingspeak with 80', but this is not working. Wikipedia works, but thingspeak not. Thingspeak only works with port 80 AND with ``WiFiClient.h``. But when i do so, i can't connect to wikipedia of course.

supersjimmie, 2017-11-06:
Use 2 separaste clients, like:
```
#include <WiFiClient.h>
#include <WiFiClientSecure.h>
WiFiClient client;
WiFiClientSecure clientsecure;
```
Then use the ```client``` for Thingspeak and ```clientsecure``` for wikipedia.

Staubgeborener, 2017-11-06:
@supersjimmie I thought about the same. But this also won't work. I can get a connection to wikipedia, but not to thingspeak. See updated code.

Staubgeborener, 2017-11-06:
@supersjimmie Oh man. I thought about the same, but it was not working, cause i made this tippos:
``WiFiClientSecure client;
WiFiClientSecure clientsecure;``

Thanks. That was quite simple, when you can write properly.

closed 2017-11-06 by Staubgeborener