esp8266/Arduino #3804: softAP disappears

Aminabdiyan, 2017-11-06:
#### Hardware
Hardware:			ESP-12f
Core Version:      	2.3.0

### Description

ESP8266 access point disappears after a while. Can I check that access point exists in loop? I use WiFi.getMode() ,WiFi.softAPIP() but not changed after disappearing AP. 

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		1MB 
CPU Frequency:		80Mhz  //I also checked 160Mhz
Flash Mode:			qio //I also checked dio
Flash Frequency:	80Mhz
Upload Using:	      SERIAL
Reset Method:		ck

I choose different IDE settings(Node mcu,Adafruit Huzzah ESP8266) but problem persist. 
### Sketch(summary)
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <FS.h>
#include "EEPROM.h"
#include "_74HC595.h"
#include "MD5.h"
#include "RemoteReceiver2.h"
#include <ESP8266mDNS.h>  //I don't use this library in AP mode
extern "C" {
#include "user_interface.h"
}
ESP8266WebServer* server;
void setup() {
...
IPAddress apIP(192, 168, 3, 1);
WiFi.mode(WIFI_AP);
WiFi.persistent(true);
WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
WiFi.softAP(SSIDName,WiFiAPPSK,6); // I checked WiFi.softAP(SSIDName,WiFiAPPSK) too. 
server=new ESP8266WebServer(80);
server->on("/r", HTTP_POST, []() {
...
});
server->onNotFound([]() {
    if (!handleFileRead(server->uri()))
      server->send(404, "text/plain", "FileNotFound");
  }); 
server->begin();
Udp.begin(2390);
...
}
void loop() {
...
server->handleClient();
packetSize = Udp.parsePacket();
if (packetSize) 
{
packetBuffer[0] = 0; //buffer to hold incoming packet
packetBuffer[1] = 0;
Udp.beginPacket(Udp.remoteIP(), Udp.remotePort());
 Udp.write("h");
}
...
}

### Debug Messages

```
I delete board esp8266 and reinstall it today after that when I enable debug on Serial WDT resets(I don't know why)

there is no poison after the block. Expected poison address: 0x3fff2ab4, actual data: 0xaf 0x0 0xa5 0xa5
block start: 3fff2a04

Panic C:\Users\amin_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc\umm_malloc.c:853 check_poison_block

ctx: sys 
sp: 3ffffd40 end: 3fffffb0 offset: 01b0

>>>stack>>>
3ffffef0:  5ffffe00 5ffffe00 3ffefb28 40100334  
3fffff00:  4a00361f 00000ca0 3fff29f8 4010044f  
3fffff10:  4f4cbf71 ffffffff 3fff0420 4010054f  
3fffff20:  ffffffff 3fff1a38 3fff1dfc 401008d9  
3fffff30:  ffffffff ffffffff 00007bc6 40106f8c  
3fffff40:  40105246 00007bc4 ffffffff 402331e5  
3fffff50:  401060b4 00000000 00007bc4 00000000  
3fffff60:  40228d0c 3ffefae8 3ffefb10 60000600  
3fffff70:  021040fa 3ffefb10 3ffefae8 40228d19  
3fffff80:  40228d5e 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffefb30 3fffdad0 3fff0500 40211253  
3fffffa0:  40000f49 97b9bbff 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:1, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

```
Sorry for my English.

hamsafar, 2017-11-06:
Hi,I have the same problem too...

Aminabdiyan, 2017-11-09:
when I remove EEPROM.commit() in loop, ESP works with no problem, but I want to use EEPROM how solve this problem?
 EEPROM.commit() function erase and write 4096 bytes of data, can I decrease it? 
Has any problem if I don't erase 4095 byte of data and only write it? or since my EEPROM data is very important and in rare cases EEPROM.commit() may only erase data without writing again, how can change the code that if EEPROM.commit() erase data surely write data. like  TRANSACTION sql server code:
BEGIN { TRAN | TRANSACTION } 
    [ { transaction_name | @tran_name_variable } 
      [ WITH MARK [ 'description' ] ] 
    ] 
[ ; ]

devyte, 2017-11-11:
@Aminabdiyan it is difficult to figure out exactly what is wrong when your sketch is incomplete. Please edit your post and post a Minimal Complete Verifiable Example.
Also, 2.3 is very old, please try 2.4-rc2 or better yet, latest git.
Finally, I see that you're using UDP. Please try building with lwip2 over latest git (option in the menu in Arduino IDE). There are known issues with lwip and old lwip1.4.

tablatronix, 2017-11-19:
does it do it when you remove your eeprom code?
Did you do full erase with esptool ?

Aminabdiyan, 2017-11-19:
@devyte I solve this problem with doing EEPROM.commit() every five minutes,
@tablatronix yes, before I did full erase but didn't solve the problem

tablatronix, 2017-11-19:
Just asking because the esp eraseconfig is bugged in stable, so you must use esptool instead

Aminabdiyan, 2017-11-19:
@tablatronix thanks, I use https://github.com/kentaylor/EraseEsp8266Flash to erase flash

devyte, 2018-09-06:
@Aminabdiyan is this issue still valid?

devyte, 2018-11-12:
Closing due to lack of feedback in a long time.

closed 2018-11-12 by devyte