esp8266/Arduino #3805: Client.connect fails to 192.168.4.1 of another ESP8266

Jsolo1, 2017-11-06:
### Basic Infos

#### Hardware
Hardware:			NodeMCU v3
Core Version:      	2.4.0-rc2

### Description

I'm using one master ESP8266 to program others running the Wifimanager. The Master esp connects to the AP created by others and loads the SSID and password. At least that's the theory. In practice I can get the master to connect to the AP but then the GET request to load the SSID and pass fails. It fails to connect even, I think it might because of using an IP instead of a Domain name.
Tried with both HTTPrequest and Wificlient but none of them seem to work for the application.

### Settings in IDE

Module:  			NodeMCU
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			?qio?
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch

```cpp
#include <ESP8266WiFi.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>

#include <ESP8266HTTPClient.h>

#define pushbuttonPin 0

//WiFiClient client;
HTTPClient http;

void ICACHE_RAM_ATTR ButtonPushed();    //to keep the function handler in IRAM

int portal = 0;
long pushedlast = 0;
const char* host = "192.168.4.1";

void setup() {
  Serial.begin(115200);

  WiFi.begin("WIFI-AP", "Password");
  

  pinMode(pushbuttonPin, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(pushbuttonPin), ButtonPushed, CHANGE);

  
}

void loop() {
  
  if (WiFi.status() == 3){
    Serial.println("Connected to network");
    http.begin("http://192.168.4.1/wifisave?s=NEW-SSID&p=newpass");
    int httpCode = http.GET();
    Serial.println(httpCode);
    if (httpCode > 0)
    //if (client.connect(host, 80))
    {
      Serial.println("Connected to host");
      String payload = http.getString();
      Serial.println(payload);
      http.end();
      // we are connected to the host!
      //client.print(String("GET /wifisave?s=NEW-SSID&p=newpass") + " HTTP/1.1\r\n" +
      //client.print(String("GET /") + " HTTP/1.1\r\n" +
             //"Host: " + host + "\r\n" +
             //"Connection: close\r\n" +
             ////"User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/61.0.3163.100 Safari/537.36" +
             ////"Accept: text/html" +
             //"\r\n"
            //);
    }
    else
    {
      // connection failure
      Serial.println("Failure to connect to host");
    }
  }
  else{
    Serial.println("Waiting for connection");  
  }
  

  delay(1000);

}
```

### Debug Messages

```
messages here
```

rick-bp, 2017-11-09:
I'm having the same problem, but I think this comes from bellow versions...

I have a project that have one ESP8266 as AP and others ESP8266 as STA.
As it is obvious, the ESP stations conect on ESP AP.
When the station start and conect to AP, it send a hello with some parameters behind httpClient.

The problem is: The hello() function works perfectly at the first time that is called.
Next call of hello(), I receive: -1 Connection Refused

Somebody have a tip, trick, a magic, or who know the answer for this?

Look this:

`
#include "Arduino.h"
#include <EEPROM.h>
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <FS.h>
#include <ESPAsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include "AsyncJson.h"
#include "ArduinoJson.h"
#include <StreamString.h>
#include "TimeLib.h"
#include <ESP8266HTTPClient.h>
#include <stdio.h>

#include "lib.h"
#include "version.h"
#include "wifihwtypes.h"

.....

void WiFiDCC::hello() {
#ifndef RELEASE
  Serial.printf_P( pgm_INFunction, "hello()", ESP.getFreeHeap() );
#endif

  HTTPClient http;
  String payload;
  String url;

  url = "http://" + wifi.gateway.toString() + "/gsr?qt=run&q=hello&hwType=" + hw.HW.Type + "&hwSN=" + lib.urlencode(searchSN()) + "&IP=" + wifi.IP.toString() + "&hwName=" + lib.urlencode(hw.hwName);

#ifndef RELEASE
  Serial.println(url);
#endif

  if ( WiFi.status() == WL_CONNECTED ) {
    http.begin(url);
    int httpCode = http.GET();
#ifndef RELEASE
    if (httpCode < 0) {
      Serial.printf("\n[HTTP] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
    }
#endif

    payload = http.getString();
    http.end();

#ifndef RELEASE
    Serial.print("payload: ");
    Serial.println( payload );
#endif

    DynamicJsonBuffer jsonBuffer(32);
    JsonObject& root = jsonBuffer.parseObject(payload);
    IPWiFiStation = lib.strToIP(root["ip"]);
  } else {
#ifndef RELEASE
    Serial.println( F("Desconectado enquanto iniciava hello()") );
#endif
  }

  payload = String();
  url = String();

#ifndef RELEASE
  Serial.printf_P( pgm_OUTFunction, "hello()", ESP.getFreeHeap() );
#endif
}
`

devyte, 2017-11-12:
@Jsolo1 your sketch seems to be incomplete.
Off the top of my head, I see this:
in setup():
`  WiFi.begin("WIFI-AP", "Password");
`
no static IP configured, no waiting for DHCP response, no WiFi event setup.

in loop():
`  if (WiFi.status() == 3){
`
which checks if WiFi is connected. This doesn't mean that a DHCP response has been received, nor that an IP address is currently leased.
Therefore, the code that follows can attempt to execute when a network is not yet setup, and hence the connection fails.

I strongly suggest looking at the WiFi events. Example: on GotIP event set a flag, on Connection lost event unset the flag, and in loop() execute your code based on that flag.

Closing due to user error.

closed 2017-11-12 by devyte

Jsolo1, 2017-11-13:
Hi, thanks for the answers.
I've added this line to check if there is an IP assigned:

`Serial.println(WiFi.localIP());`

And there is: "192.168.4.2"

Still the connection fails.

Jsolo1, 2017-11-13:
Digging a bit further:

[HTTP-Client][begin] url: http://192.168.4.1/wifisave?s=NEWAP&p=newpass
[HTTP-Client][begin] host: 192.168.4.1 port: 80 url: /wifisave?s=NEWAP&p=newpass
[HTTP-Client] failed connect to 192.168.4.1:80
[HTTP-Client][returnError] error(-1): connection refused

So it is somehow refusing to connect.

Jsolo1, 2017-11-15:
Can you please reopen this issue?

jcluq, 2021-02-24:
I am also looking for help with this.