esp8266/Arduino #3806: Typo in reent.h

retsifp, 2017-11-06:
Sorry, too lazy to fill out everything ;-)

With the latest GIT version and the line `#include "FS.h"` in my sketch (Updater.h:17), the following error occurs during compilation:

> In file included from /home/thomas/git/Arduino_Projects/hardware/esp8266com/esp8266/tools/sdk/libc/xtensa-lx106-elf/include/wchar.h:6:0,
                 from /home/thomas/git/Arduino_Projects/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/cwchar:44,
                 from /home/thomas/git/Arduino_Projects/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/bits/postypes.h:40,
                 from /home/thomas/git/Arduino_Projects/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/iosfwd:40,
                 from /home/thomas/git/Arduino_Projects/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/memory:72,
                 from /home/thomas/git/Arduino_Projects/hardware/esp8266com/esp8266/cores/esp8266/FS.h:24,
                 from sketch/Updater.h:17:
/home/thomas/git/Arduino_Projects/hardware/esp8266com/esp8266/tools/sdk/libc/xtensa-lx106-elf/include/sys/reent.h:458:10: error: #if with no expression
 #if DEBUG
          ^
exit status 1

When replacing the [line](https://github.com/esp8266/Arduino/blob/master/tools/sdk/libc/xtensa-lx106-elf/include/sys/reent.h#L458) by e.g. `#ifdef DEBUG`, everything works.

I am not sure why this bug hasn't been filed yet, maybe I'm doing something very special?
Updater.h is btw. included from a cpp file.

igrr, 2017-12-27:
Indeed, you must be doing something special, or have something unique about your setup. I am not able to reproduce this manually with the latest git version, and CI builds also pass successfully (there are a few .cpp files which include FS.h).

If this is still an issue, could you please describe your setup in more detail?
FWIW, i think the line in sys/reent.h is perfectly valid... unless someone somewhere added a -DDEBUG flag into CFLAGS. Which i don't see being done in platform.txt. Perhaps something is set in your shell environment?

retsifp, 2017-12-28:
You are right. I have a
`#define DEBUG` in a settings-file and I use it for some debug-output.

I think I will have to rename it... ;-)

closed 2017-12-28 by retsifp

electronicsguy, 2018-01-30:
@retsifp @igrr Same issue! If we have our own #define DEBUG, this line gives an error. Ivan, the statement itself may be correct syntax, but wouldn't it be better to change it to #ifdef (because that's what youre' really going for)?