esp8266/Arduino #3807: Cannot upload my sketch.

crazyhorse961, 2017-11-07:
### Basic Infos

#### Hardware
Hardware:			?I'm not aware of it, i bought it here tho: https://goo.gl/zs32sC?
Core Version:      	?Not aware of it?

### Description

Cannot upload my sketch.

### Settings in IDE

Module:  			?Olimex MOD-WIFI-ESP8266?
Flash Size: 		?1MB?
CPU Frequency:	?80Mhz?
Flash Frequency:	?40Mhz?
Upload Using:		?SERIAL?
Upload Speed:        ?9600?

### Debug Messages
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_open failed

I got drivers working, the esp8266 connects successfully to my Windows 10 pc.
I also pressed the RST button so the esp8266 is totally empty.

caverna, 2017-11-08:
Hi @crazyhorse961, please look at the issue #3547.
There I realize two things: one is that the MCU programmer is more tolerant to errors, second you don't need a 30A power source, but you must have very good capacitors in VCC pins!

WereCatf, 2017-11-08:
@crazyhorse961 The board you linked to is not an Olimex-board. Change your IDE-settings to NodeMCU 0.9 and try again.

crazyhorse961, 2017-11-08:
Hello @WereCatf, i changed it to NodeMCU 0.9 and i get the same output.

crazyhorse961, 2017-11-08:
Updates, now the blue LED turns on, but i still cannot find the WiFi network.
I tried to send commands with RealTerm but i don't receive any output, if this can help.

porkyneal, 2017-11-09:
Have you set the correct baud rate in RealTerm to match that of the ESP?

Pablo2048, 2017-11-09:
Guys but this is known problem - raise your upload speed to something higher, NOT 9600bd. I suggest >57600 ...

crazyhorse961, 2017-11-09:
@Pablo2048 this ESP's recommended bd is 9600, i wont increase it, i also tried running it with 57600 but same thing.
@porkyneal Yes.

Pablo2048, 2017-11-09:
@crazyhorse961 huh? Where from you got this (IMHO stupid) speed recommendation? Do you imagine how much time take for example upload 3MB SPIFFS image with this speed?

crazyhorse961, 2017-11-09:
@Pablo2048 it's a text behind my esp8266 on connectors.
It litterally says "Use 9600bps baud rate"

Pablo2048, 2017-11-09:
But this is for Lua firmware right? Not for Arduino one...

crazyhorse961, 2017-11-09:
I'm not aware.. ill try with 115200.

Pablo2048, 2017-11-09:
Ok, it seems that you are a noob with ESP so here are my suggestions:
1. this is not right place to ask such questions (use the forum) - develo probably closes this thread soon...
2. set your serial speed to 115200 and try to upload blink (or another simple sketch)
3. if this fail try to change reset method (i'm using nodemcu) and try to upload sketch again

crazyhorse961, 2017-11-09:
Yep, its litterally my first project with these cool things, and now it's also working! :)
Thanks to everyone

closed 2017-11-09 by crazyhorse961