esp8266/Arduino #3810: ESP8266 crashes when logging plain request body

mobsome, 2017-11-08:
https://github.com/esp8266/Arduino/blob/303a71deeaac71cf69c2fc5d9aa4215a7950466f/libraries/ESP8266WebServer/src/Parsing.cpp#L206

When log is enabled and request with long body is logged ESP crashes. Prints should be limited.
Tested on WeMos ESP-12.

[crash.txt](https://github.com/esp8266/Arduino/files/1455008/crash.txt)

devyte, 2017-11-10:
@mobsome no basic info, no IDE settings, no MCVE sketch to reproduce the issue.
What are the settings for the serial port (speed)?
What is the decoded output of the stack dump?
Without some context, it is unlikely that this issue will be handled.

tgriffith42, 2017-12-01:
I found that regardless of log... simply trying to display a large web page crashes on the ESP8266 after the 4th or 5th page load/refresh.  I had a string of about 5,000 characters that I split into 3 different client.print(); commands.  
Mainly because I was loading images from google drive - which have long URLs.  
I got rid of all the images and got rid of anything not necessary, and got my web page down to less than 2,000 characters, and now the ESP8266 no longer crashes.

The ultimate solution would be to host the web page on a computer, and add an iFrame and JavaScript with form.submit(); to call the ESP8266 - that way ESP8266 is simply used for processing commands - and not also rendering a web page.

mobsome, 2017-12-01:
Please remember that ESP can work as access point. In such configuration any app that is installed on device connected to ESP wifi treats it as a gateway and can send any kind of data of any size over ESP.
That's way webserver code should limit or split data that are being printed.

devyte, 2017-12-02:
@tgriffith42 what you describe is something completely different. I suggest trying latest git with lwip2, and if that doesn't work, try ESPAsyncWebserver instead of the standard one.

earlephilhower, 2019-01-16:
Closing as no MCVE, no clear description of problem.  Nothing we can do.

closed 2019-01-16 by earlephilhower