esp8266/Arduino #3811: ESPhttpUpdate.update ignores 301 and 302 redirect http status code

mahees, 2017-11-09:
Hi everyone,

Im working on a lambda function as the update hander. It uses the HTTP_X_ESP8266_STA_MAC and HTTP_X_ESP8266_VERSION header to determine if the esp version is up to date. if it is, returns 304 (not modified). If not, then i'm currently trying to return a 301 redirect to the location of the lastest bin file to be installed from an s3 bucket.

When i try with the 301, the ESPhttpUpdate.update function sees this as a failure and bad response code. 

Would it be possible to update ESPhttpUpdate.update to follow 301 and 302s?

thanks much

davisonja, 2017-11-09:
Yes, it would 😊

devyte, 2017-11-10:
@mahees I don't quite understand the request. Could you please provide an example/sketch/diagram of what you want to do?

davisonja, 2017-11-11:
I believe the request is that the ESPhttpUpdate.update routine will act on a 301 and 302 redirect codes. At present it expects either a 200 code (which is a valid update to apply) or a 304 code (which means no update available). This request desires 301 and 302 to also be valid responses, which trigger an update using the referenced URL as the binary.

davisonja, 2017-11-11:
If I'm right, I'm happy to do that :)

mahees, 2017-11-11:
hi @devyte This request is to follow http / https redirects (301 and 302 status codes with the Location header)

t_httpUpdate_return ret = ESPhttpUpdate.update(https://<some-script-that-generates-s3-url>, SKETCH_NAME, [LAMBDA_SSL_SIGNATURE, S3_SSL_SIGNATURE]);

https://<some-script-that-generates-s3-url> could be a php or nodejs script that returns a 301 status code with the location header to the exact link to the binary file (in my case to s3). 
eg.
HTTP/1.1 301 Moved Permanently
Location: https://some-bucket-name.s3.amazonaws.com/httpUpdate.ino.bin

s3 is for simple file storage where the updated sketch binary is stored

hi @davisonja awesome! yes you are spot on. The complexity i see here are the different ssl / https sha1 signatures for lambda (api gateway) and s3 urls. Can you advise if you will be working on this? Would be epic awesome!

Thanks!

mahees, 2017-11-17:
hey everyone, 

here is a drawing of what i am trying to do
https://docs.google.com/drawings/d/11_g4em223vnbjms_Kr3ZcWv2Cl0fKIqF2UdagAzVmn4/edit?usp=sharing

Here is my hacky work around so far...
https://gist.github.com/mahees/bfd83c4230b5152d013f242d8e72f917#file-esp-http-update-ota-s3-lambda-cpp-L41

However, since s3 is just a fileserver, its not able to generate the `x-MD5` header so i'm getting this error `HTTP_UPDATE_FAILD Error (-106): Verify bin header failed`

mahees, 2017-11-17:
[image]

mahees, 2017-11-17:
Im not a cpp dev, so i'm hoping someone can help me with this (perhaps should be a different issue request)

https://github.com/esp8266/Arduino/commit/95c8152d26ae245ba89d27272852722e872e831c

thanks!

jlujan-na, 2017-11-21:
> However, since s3 is just a fileserver, its not able to generate the x-MD5 header

S3 in practice sets the ETag header to the MD5 of the file. There is no requirement that the ETag header be the MD5 of the file, but many implementations use this. Nginx uses last-modified-time + content length, so not universal. 

Using the ETag=MD5, client sending "If-None-Match" request header, and a fixed url for the firmware "yourserver/latest.bin" would allow for expected behavior (200 on change / 304 on no update) out of S3 with no other server.

As for the feature request, being able to override this header ('x-MD5') in a #define would be nice.

devyte, 2019-11-20:
@mahees a lot has changed since this was reported. Is this issue still valid?

mahees-hyuna, 2019-11-20:
No. We can close this. Thank you!

closed 2019-11-20 by devyte