esp8266/Arduino #3812: Unable to use http.setAuthorization for ESPhttpUpdate

pjunni, 2017-11-09:
I did not find a way to use http.setAuthorization for ESPhttpUpdate, but I solved the issue by adding the line http.setAuthorization("username","password"); on line 181 in https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266httpUpdate/src/ESP8266httpUpdate.cpp. Is there a more "correct" way to do this or could there be an overloaded handleUpdate function that takes these "username" and "password" and pass them into http.setAuthorization("username","password");?

devyte, 2017-11-10:
@pjunni if I understand the request correctly, what you want to do is update from a webserver that requires user/pass. Is that correct?

pjunni, 2017-11-13:
That's correct.

pjunni, 2017-11-23:
@devyte Would you have any suggestions on this matter?

jatayu86, 2017-12-11:
I face the similar issue,could someone please let us know how to add http authorization headers for ESPHttpUpdate?

ngashok, 2019-04-26:
Facing same issue. How to do OTA from https server having authorisation with username/password.

Swiftnesses, 2019-08-19:
Was this ever solved, I have the same issue?

drhideg, 2020-04-10:
If you are still interested, would you give it a try this PR? #7190

earlephilhower, 2021-01-08:
Fixed via the mentioned PR (#7190).  Closing.

closed 2021-01-08 by earlephilhower