esp8266/Arduino #3813: core_esp8266_features.h: No such file or directory

Schawen, 2017-11-09:
Hello,

I tried to compile and upload this WebSocket Demo:
https://tttapa.github.io/ESP8266/Chap14%20-%20WebSocket.html 

to the connected NodeMCU 1.0 (ESP-12E Module) , but get this error:
core_esp8266_features.h: No such file or directory

I've than updated "Arduino core for ESP8266 WiFi chip" following the instructions from:
https://github.com/esp8266/Arduino#using-git-version-

Using git version
-------------------
Linux build status
    Install Arduino 1.8.2 from the Arduino website.
    Go to Arduino directory
    Clone this repository into hardware/esp8266com/esp8266 directory (...)
cd hardware
mkdir esp8266com
cd esp8266com
git clone https://github.com/esp8266/Arduino.git esp8266
    Download binary tools (you need Python 2.7)
cd esp8266/tools
python get.py

Everything worked fine until I entered the last command ( python get.py ), because python is not in the Path:

D:\Dokumente\Arduino\hardware\esp8266com\esp8266\tools>d:\Programme\Python2.7\python.exe get.py
Platform: i686-mingw32
Tool esptool-0.4.12-win32.zip already downloaded
Extracting dist/esptool-0.4.12-win32.zip
Renaming esptool-0.4.12-win32/ to esptool
Tool win32-xtensa-lx106-elf-gb404fb9-2.tar.gz already downloaded
Extracting dist/win32-xtensa-lx106-elf-gb404fb9-2.tar.gz
Tool mkspiffs-0.2.0-no_magic_length-windows.zip already downloaded
Extracting dist/mkspiffs-0.2.0-no_magic_length-windows.zip
Renaming mkspiffs-0.2.0-no_magic_length-windows/ to mkspiffs

D:\Dokumente\Arduino\hardware\esp8266com\esp8266\tools>

I thought that this should have fixed the problem, but I further on receive the same error.
Here is the complete output from compiling the Demo code:

Arduino: 1.8.3 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\MiKo\AppData\Local\Arduino15\packages -hardware D:\Dokumente\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\MiKo\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Dokumente\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:CpuFrequency=80,UploadTool=esptool,UploadSpeed=115200,FlashSize=4M3M -ide-version=10803 -build-path C:\Users\MiKo\AppData\Local\Temp\arduino_build_408119 -warnings=none -build-cache C:\Users\MiKo\AppData\Local\Temp\arduino_cache_402542 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=C:\Users\MiKo\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.6 -prefs=runtime.tools.mkspiffs.path=C:\Users\MiKo\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.1.2 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\MiKo\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -verbose D:\Dokumente\Arduino\ESP8266\A-WebSocket_LED_control\A-WebSocket_LED_control.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\MiKo\AppData\Local\Arduino15\packages -hardware D:\Dokumente\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\MiKo\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Dokumente\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:CpuFrequency=80,UploadTool=esptool,UploadSpeed=115200,FlashSize=4M3M -ide-version=10803 -build-path C:\Users\MiKo\AppData\Local\Temp\arduino_build_408119 -warnings=none -build-cache C:\Users\MiKo\AppData\Local\Temp\arduino_cache_402542 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=C:\Users\MiKo\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.6 -prefs=runtime.tools.mkspiffs.path=C:\Users\MiKo\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.1.2 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\MiKo\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -verbose D:\Dokumente\Arduino\ESP8266\A-WebSocket_LED_control\A-WebSocket_LED_control.ino
Using board 'nodemcuv2' from platform in folder: C:\Users\MiKo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0
Using core 'esp8266' from platform in folder: C:\Users\MiKo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0
Detecting libraries used...
"C:\Users\MiKo\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\MiKo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10803 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266 "-IC:\Users\MiKo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266" "-IC:\Users\MiKo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\variants\nodemcu" "C:\Users\MiKo\AppData\Local\Temp\arduino_build_408119\sketch\A-WebSocket_LED_control.ino.cpp" -o "nul"
"C:\Users\MiKo\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\MiKo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10803 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266 "-IC:\Users\MiKo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266" "-IC:\Users\MiKo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\variants\nodemcu" "-IC:\Users\MiKo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src" "C:\Users\MiKo\AppData\Local\Temp\arduino_build_408119\sketch\A-WebSocket_LED_control.ino.cpp" -o "nul"
"C:\Users\MiKo\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\MiKo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10803 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266 "-IC:\Users\MiKo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266" "-IC:\Users\MiKo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\variants\nodemcu" "-IC:\Users\MiKo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src" "-IC:\Users\MiKo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ArduinoOTA" "C:\Users\MiKo\AppData\Local\Temp\arduino_build_408119\sketch\A-WebSocket_LED_control.ino.cpp" -o "nul"
"C:\Users\MiKo\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\MiKo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10803 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266 "-IC:\Users\MiKo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266" "-IC:\Users\MiKo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\variants\nodemcu" "-IC:\Users\MiKo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src" "-IC:\Users\MiKo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ArduinoOTA" "-IC:\Users\MiKo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WebServer\src" "C:\Users\MiKo\AppData\Local\Temp\arduino_build_408119\sketch\A-WebSocket_LED_control.ino.cpp" -o "nul"
[... cut ...]
Bibliothek ESP8266WebServer in Version 1.0 im Ordner: C:\Users\MiKo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WebServer  wird verwendet
Bibliothek ESP8266mDNS im Ordner: C:\Users\MiKo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266mDNS (legacy) wird verwendet
Bibliothek arduinoWebSockets-master in Version 2.0.10 im Ordner: D:\Dokumente\Arduino\libraries\arduinoWebSockets-master  wird verwendet
exit status 1
Fehler beim Kompilieren für das Board NodeMCU 1.0 (ESP-12E Module).

Hope somebody can help ?

5chufti, 2017-11-09:
see here for instructions how to make clean install w/o python
(or just dld get.exe to get you going)
http://www.esp8266.com/viewtopic.php?f=26&t=14222

devyte, 2017-11-10:
@Schawen this is an issue tracker, not a community forum. Please discuss at esp8266.com or stackoverflow.
I just did a clean install under Ubuntu, and I didn't encounter a problem.
Closing as can't reproduce, and off-topic.

closed 2017-11-10 by devyte