esp8266/Arduino #3814: gettimeofday incorrect value for tv_usec

BrandonLWhite, 2017-11-09:
The current implementation _gettimeofday_r is incorrectly setting the tv_usec field.

https://github.com/esp8266/Arduino/blob/master/cores/esp8266/time.c#L104

According to http://www.gnu.org/software/libc/manual/html_node/Elapsed-Time.html

tv_usec should be "This is the rest of the elapsed time (a fraction of a second), represented as the number of microseconds. It is always less than one million."

The current code is simply doing `tp->tv_usec = micros();`.  This leads to the wall clock time advancing forward too quickly.  I am certainly aware of the caveats of tracking wallclock (ie CLOCK_REALTIME) in ESP8266, but this is just making it far worse.

I have some ideas on how to square this away once and for all (crossing my fingers).  I will try to follow-up with additional info or possibly a pull request.

5chufti, 2017-11-09:
so basically it boils down to micros() - millis()*1000
or simply micros() % 1000000 if tv_sec is micros()/1000000

BrandonLWhite, 2017-11-09:
I think you have the right idea, but the problem with using micros directly like that is that it is only 32-bit.  This could all be gravy if we had a `uint64_t micros64()` version.

Something like this is a step in the right direction:

```
uint64_t elapsed_ms = s_bootTime + millis();
tp->tv_sec  = elapsed_ms / 1000;
tp->tv_usec = (elapsed_ms % 1000) * 1000;
```

I think that would work in a pinch.  Unfortunately this would still be bugged over the long haul because s_bootTime is never updated and millis() overflows after 2^32 ms.

5chufti, 2017-11-10:
@d-a-v : I think this is a topic along #1679 ...

PR #3830 (2017-11-14): Added micros64 and used to fix and improve gettimeofday.
This resolves #3814 and adds a `micros64()` global function that is very useful.

There is certainly more that can be done to improve wallclock time-keeping, but this is a step in the right direction I feel.  This allows my application to have a reasonably decent current timestamp.

This now works well enough for me, when it previously was wrong:
```
auto now_ms = std::chrono::duration_cast<std::chrono::milliseconds>(std::chrono::system_clock::now().time_since_epoch()).count();
```

I purposely did not touch `clock_gettime()` or `time()` [... cut]
closed 2017-11-14 by igrr