esp8266/Arduino #3815: Connecting client issue when sending HTTP request

Cpt-Hook, 2017-11-09:
### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description
Hi, I want to send a request to ESP8266 and it works properly, unless I send a request which is as follows:
<any http method> <any message> HTTP/1.1
for example:
GET /turnonled HTTP/1.1
When I send this request the code exits through client timeout. If i send any other request, that does not contain GET in the beggining (or POST or any HTTP method) and HTTP/1.1 it works fine.
what is the issue? kind of strange really
### Sketch

```cpp

unsigned long time = millis();

while(!client.available()){
    if(millis() - time > 2500){
      Serial.println("client timeout");
      client.flush();
      return;
    }
  }

String req = client.readString();
```

devyte, 2017-11-10:
@Cpt-Hook no basic info filled in, example is incomplete. Please provide the info requested in the issue template, and an MCVE sketch, or this issue will be closed.
Is this issue valid with latest git?

Cpt-Hook, 2017-11-12:
#### Hardware
Hardware:		ESP8266MOD
Core Version:      	2.1.0-rc2

### Description

I am trying to send a request to my ESP. The handleServer function is called repeatedly from my loop(). 
This code works very well for most requests. 

But when I send a request, which is only an incomplete http request (for example I send "GET" or "POST" or "GET / HTT") code always ends up in client timeout (as marked in code). The function client.available() always returns 0, even though I actually sent something and bytes are ready to be read. 

For example when I send "hello" or "this is a request" or  "GET / HTTP1.1\r\n" it works OK. I dont understand. Does client.available() function return 0 if it finds incomplete HTTP request? If so how can I make it not behave like this? I want to be able to send arbitrary request independent of protocol used.

I wrote a simple Java program to send these arbitrary requests. Requests sent from the browser work, because they are complete and valid HTTP requests.
### Settings in IDE

Module:  			 Wemos D1 board
Flash Size: 		4MB
CPU Frequency:	80Mhz
Upload Using:		SERIAL
### Sketch

```cpp
void handleServer(){ //this function is called repeatedly
  // Check if a client has connected
  WiFiClient client = server.available();
  if (!client) {
    delay(1);
    return;
  }
  Serial.print("New client: ");
  Serial.println(client.remoteIP().toString());
  
  unsigned long time = millis();
  
  while(!client.available()){
    if(millis() - time > 1500){
      Serial.println();
      Serial.println("client timeout"); //code ends up here if incomplete http req is sent
      Serial.println();
      client.flush();
      return;
    }
    ;
  }

  String req = "";
  while(client.available()){
    req += client.readStringUntil('\r');
  }
  Serial.println("The request is:");
  Serial.println(req);
}
```

devyte, 2018-07-13:
There have been several fixes merged since this was opened, especially in lwip glue, which is relevant in the posted code. 
Considering heavy recent testing to the lwip glue code, I am closing this. Please open a new issue if you think this problem is not fixed.

closed 2018-07-13 by devyte