esp8266/Arduino #3816: Promiscuous mode prevent to send UDP packet

fralomb, 2017-11-09:
### Description
Basically i want to enable the promiscuous mode on the nodeMCU and sometimes send some UDP packets in the network to a remote server. I've tried a simple example in which the module (after having made the connection to the router) sends these UDP packets to the server and on the other side the server receives them correctly. But if i enable the promiscuous mode, by registering the callback through the wifi_set_promiscuous_rx_cb(..) function, then my UDP server no longer receives any packet from the ESP module. Moreover, by monitoring the network traffic with Wireshark no UDP packet actually does not pass any packets. 
The callback registered for sniffing the packets simply ignore packets that are not a "Probe Request", but i'm wondering how this is related with the send of an UDP packet, that is connectionless and so it is not expected the reception of any kind of message. I'm a bit confused, someone can help me?

#### Hardware
Hardware:			ESP-12E
Core Version:      	        2.3.0

### Settings in IDE
Module:  			NodeMCU 1.0 (ESP-12E module)
Flash Size: 		        4MB
CPU Frequency:		80Mhz
Upload Using:		SERIAL

d-a-v, 2017-11-10:
I don't know about this SDK function.
If you don't find any help about it, [here](https://github.com/esp8266/Arduino/issues/2330#issuecomment-342035256) another simple way to enable a sort of promiscuous mode while not stopping the normal behaviour of any sketch (receive through the regular API is still enabled).

devyte, 2017-11-10:
@fralomb no MCVE sketch provided, no details on how to reproduce the issue.
Your question is about cross-functionality with an SDK function, which is not really supported in the features offered in this repo. In addition, it is a request for help, which is off-topic. This is an issue tracker, not a support forum.
Please refer to a community forum for discussion. If you find an issue in the code hosted in this repo, please open a new issue, and fill in the requested fields.
Closing per #3655 .

closed 2017-11-10 by devyte