esp8266/Arduino #3817: tools/get.py CORRUPTED/dist DOWNLOADS

nemo9955, 2017-11-10:
### Basic Infos

#### Hardware
Hardware:	(uname -a)	Linux debst 4.9.0-3-686-pae #1 SMP Debian 4.9.30-2+deb9u5 (2017-09-19) i686 
Core Version:      	2.3.0

### Description

Problem description
Debian 9 on an old laptop, the script seems to be downloading corrupted/altered archive files.
Removing the SHA check, the code still fails with `not a gzip file` .
Changed source to only show needed download link 
`
Platform: i686-pc-linux-gnu
https://github.com/igrr/esptool-ck/releases/download/0.4.12/esptool-0.4.12-linux32.tar.gz
https://github.com/esp8266/Arduino/releases/download/2.3.0/linux32-xtensa-lx106-elf.tar.gz
https://github.com/igrr/mkspiffs/releases/download/0.2.0/mkspiffs-0.2.0-no_magic_length-linux32.tar.gz
`
I downloaded them and put them in /dist, reverted source of get.py and it worked.

### Debug Messages

```

======= original stack trace =============

Platform: i686-pc-linux-gnu
Downloading esptool-0.4.9-linux32.tar.gz
Done
Hash mismatch for dist/esptool-0.4.9-linux32.tar.gz, delete the file and try again
Traceback (most recent call last):
  File "./get.py", line 116, in <module>
    get_tool(tool)
  File "./get.py", line 83, in get_tool
    raise RuntimeError()
RuntimeError

======= removing the SHA integrity check =============

Platform: i686-pc-linux-gnu
Tool esptool-0.4.12-linux32.tar.gz already downloaded
Extracting dist/esptool-0.4.12-linux32.tar.gz
Traceback (most recent call last):
  File "get.py", line 116, in <module>
    get_tool(tool)
  File "get.py", line 84, in get_tool
    unpack(local_path, '.')
  File "get.py", line 50, in unpack
    tfile = tarfile.open(filename, 'r:gz')
  File "/usr/lib/python2.7/tarfile.py", line 1693, in open
    return func(name, filemode, fileobj, **kwargs)
  File "/usr/lib/python2.7/tarfile.py", line 1751, in gzopen
    raise ReadError("not a gzip file")
tarfile.ReadError: not a gzip file

```

igrr, 2018-01-02:
Do you still have the old (corrupted) file saved somewhere to see how exactly it was corrupted? I.e. partial download, or downloaded something totally different, or something else?

nemo9955, 2018-01-02:
I will re-download them tomorrow at work on that laptop and send them.

nemo9955, 2018-01-03:
[esptool-0.4.12-linux32.tar.gz](https://github.com/esp8266/Arduino/files/1599559/esptool-0.4.12-linux32.tar.gz)

[linux32-xtensa-lx106-elf.tar.gz](https://github.com/esp8266/Arduino/files/1599562/linux32-xtensa-lx106-elf.tar.gz)

[mkspiffs-0.2.0-no_magic_length-linux32.tar.gz](https://github.com/esp8266/Arduino/files/1599569/mkspiffs-0.2.0-no_magic_length-linux32.tar.gz)

All 3 files gave ```Hash mismatch for dist/..............``` .
Pulled the new repo, renamed working /dist -> /dist_copy, ```python get.py``` , ```esptool...```, upload it , replace it with the working one , ```python get.py``` , ```Hash mismatch for dist/linux32-xtensa-lx106-elf.tar.gz, delete the file and try again``` etc.

devyte, 2018-03-02:
@nemo9955 is this issue still valid?

nemo9955, 2018-03-02:
Will test it in the following days.

devyte, 2018-07-13:
@nemo9955 did you test this? Please test with latest git, not with 2.4.1 due to several known issues in that release.

devyte, 2018-09-06:
Closing due to lack of feedback in several months.

closed 2018-09-06 by devyte

sergio-rivas, 2018-10-22:
I am getting the exact same issue when using python(tried with 2.7.14 & also 3.7.1)  on mac osx. Hash mismatch for esptools. 

Is there a workaround? (i.e. just telling me which url's to download, and which files to run after downloading?)