esp8266/Arduino #3819: SD and FS (SPIFFS) library naming collision

pfabri, 2017-11-11:
### Description

`<FS.h>` and `<SD.h>` can't both be used in a sketch because of a naming collision. The `File` type, for example, is declared by both libraries. Attempting to build the code below will throw an error:

### Sketch

```cpp

#include <Arduino.h>
#include <SD.h>
#include <FS.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
Compiling .pioenvs/d1_mini/src/main.o
In file included from src/main.cpp:8:0:

/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/FS.h:133:11: error: 'File' is already declared in this scope
using fs::File;
^
*** [.pioenvs/d1_mini/src/main.o] Error 1
```

luc-github, 2017-11-11:
Use `#define FS_NO_GLOBALS` before include  FS
so you can use fs::File for SPIFFS and File for SD

pfabri, 2017-11-11:
@luc-github Marvellous, **it works**, thank you!
For those interested in digging deeper: [#issue112 - ESPAsyncWebServer](https://github.com/me-no-dev/ESPAsyncWebServer/pull/112)

closed 2017-11-11 by pfabri

Misiu, 2018-04-27:
@luc-github I was searching for solution to use ESPAsyncWebServer and SD card. Your solution works, thank Yo. However I think it is only temporary.
can both SPIFFS and SD have namespaces? `fs::` for SPIFFS and `sd::` for SD?
Also I found interesting thread: https://github.com/esp8266/Arduino/issues/4201
Would it solve problem with namespaces? If it would be same base File class then conflict should disappear.
Please correct me if I'm wrong.

luc-github, 2018-04-27:
temporary solutions can exist for a long time ...