esp8266/Arduino #3820: Can't save or retrieve nested structs from EEPROM

ilendemli, 2017-11-11:
#### Hardware
Hardware:	        Wemos D1 Mini (ESP-12E)
Core Version:      	2.3.0

### Description

I am trying to save a nested struct but either it does not save the data, or it can't read the data.

### Settings in IDE

Module:  			NodeMCU 1.0
Flash Size: 		4MB/1MB
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	?40Mhz?
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch

```cpp
struct Credentials {
  String ssid;
  String psk;
};

struct Memory {
  ESPROM::Credentials credentials;
  
  int timestamp;
  byte lastState;
};

static ESPROM::Memory read() {
  ESPROM::Memory memory;
  EEPROM.get(0, memory);
  EEPROM.end();
  return memory;
}

static void write(ESPROM::Memory memory) {
  EEPROM.put(0, memory);
  EEPROM.commit();
  EEPROM.end();
}

void setCredentials(ESPROM::Credentials credentials) {
  ESPROM::Memory memory = ESPROM::read();
  memory.credentials = credentials;
  ESPROM::write(memory);
 }
```

devyte, 2017-11-11:
@ilendemli what you are trying to do works only with contigous objects. String is not a contigous object, it is a managed object, where its internal memory is dynamically allocated on the heap according to need.
I suggest you investigate serialization, it is something you would need to implement for your specific needs.
Closing as user error.

closed 2017-11-11 by devyte

ilendemli, 2017-11-13:
Just posting for others who may struggle with something similar. Just managed to get it working.
```c
struct Credentials {
  char ssid[24];
  char psk[16];
};

struct Memory {
  Credentials credentials;
  
  int timestamp;
  byte lastState;
};

static Memory read() {
  Memory memory;

  EEPROM.begin(1024);
  EEPROM.get(0, memory);
  EEPROM.end();
  
  return memory;
}

static void write(Memory credentials) {
  EEPROM.begin(1024);
  EEPROM.put(0, credentials);
  EEPROM.end();
}

void setCredentials(Credentials credentials) {
  Memory memory = ESPROM::read();
  memory.credentials = credentials;
  ESPROM::write(memory);
}
```
And this is how my WebServer callback looks like:
```c
VoidCallbackFunction saveCredentialsHandler = [&]() {
  String ssid = server.arg("ssid");
  String psk = server.arg("psk");

  if (sizeof(ssid) > 0) {
    server.send(200, "text/plain", "TODO: SAVED");
    server.sendHeader("Connection", "close");

    ESPROM::Credentials credentials;
    strcpy(credentials.ssid, ssid.c_str());
    strcpy(credentials.psk, psk.c_str());
    
    saveCredentialsCallback(credentials);

  } else {
    server.send(200, "text/plain", "TODO: FAIL");
    server.sendHeader("Connection", "close");
  }
};
```