esp8266/Arduino #3825: NodeMCU Interrupt not wokring (Arduino IDE)

rajendran20, 2017-11-13:
#### Hardware
Hardware:			ESP-12
Core Version:      	?2.1.0-rc2?

Hi Guys, I am new to nodeMCU and have been trying out the following code to test the interrupt. In normal sequence the LED will blink (yellow, green, red). Once interrupt is triggered by a switch, a blue LED will blink once. Now the issue is, while main sequence is running, and interrupt is triggered, the blue led turns on and stays and worst still the main sequence is still running. The blue LED is just on. 
Switch is pulled up to 3.3V when not pressed.
Rare times, it works but the blue LED will only turn off after 8s. 

Problem description

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		?4MB
CPU Frequency:	?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:	          SERIAL?
Reset Method:		nodemcu?

---------------
Sketch
--------------

int yellow = D2; // declare yellow ledPin as variable
int green = D3; // declare green ledPin as variable
int red = D4; // declare red ledPin as variable
int blue = D5; // declare blue ledPin as variable

void setup() 
{ 
  // put your setup code here, to run once:
  pinMode(yellow,OUTPUT); //configure ledPin as OUTPUT
  pinMode(green,OUTPUT); //configure ledPin as OUTPUT
  pinMode(red,OUTPUT); //configure ledPin as OUTPUT
  pinMode(blue,OUTPUT); //configure ledPin as OUTPUT
  pinMode(SW,INPUT); //configure switch as INPUT
  attachInterrupt(digitalPinToInterrupt(SW), interruptISR, FALLING);
  
} 

void loop() 
{ 
 
    digitalWrite(yellow, HIGH); // turn on yellow LED
    delay(1000); // delay 1s 
    digitalWrite(yellow, LOW); // turn off yellow LED
    digitalWrite(green, HIGH); // turn on green LED
    delay(1000); // delay 1s 
    digitalWrite(green, LOW); // turn off green LED
    digitalWrite(red, HIGH); // turn on red LED
    delay(1000); // delay 1s 
    digitalWrite(red, LOW); // turn off red LED
   
}

 void interruptISR() {
    digitalWrite(blue, HIGH); // turn on yellow LED
    delay(1000); // delay 1s 
    digitalWrite(blue, LOW); // turn off yellow LED              // Wait for two seconds (to demonstrate the active low LED)
}

coelner, 2017-11-13:
I guess an interrupt has to be an atomic operation. The delay is probably jumping to the loop() and not back to the interrupt.

devyte, 2017-11-13:
@rajendran20 your ISR has to have the ICACHE_RAM_ATTR decorator.
Also, you can't call delay() in an ISR, let alone such a long one.
Set a flag in the ISR, check it in the loop.
Finally, you don't have wifi set up anywhere, but you also don't erase the config to remove any prior sdk params. Did you erase the flash before flashing this sketch? Having such long delays in the loop doesn't play well with the wifi stack, if it's active.

If you need further help, please discuss at a community forum.
Closing as user error, and per #3655 .

closed 2017-11-13 by devyte