esp8266/Arduino #3827: linux git install of core 2.3.0, 'xtensa-lx106-elf-g++ no such file'  compile error

gobo7, 2017-11-13:
i have a functional install of the arduino ide v1.6.9 under an older version
of slackware linux.  using the instructions for git version, i have been 
working to install the esp8266 core v2.3.0.  a very abbreviated version of the
directory tree is listed below.

on compiling the esp8266 blink example, the error
```
   fork/exec /bin/xtensa-lx106-elf-g++: no such file or directory
   Error compiling for board WeMos D1 R2 & mini.
```

this makes me believe that the issue has to be a directory name is wrong 
or an error in the path.  all research on this has been related to failed 
installs under windows.  i have found nothing to help me find the error 
under linux.

does anyone see a directory problem or have any suggestions?

many thanks.

```  
.
|-- arduino
|-- arduino-builder
|-- dist
|   |-- library_index.json
|   |-- package_index.json
|   `-- package_index.json.sig
|-- examples
|-- hardware
|   |-- arduino
|   |-- esp8266com
|   |   `-- esp8266
|   |       |-- ISSUE_TEMPLATE.md
|   |       |-- LICENSE
|   |       |-- README.md
|   |       |-- boards.txt
|   |       |-- bootloaders
|   |       |-- cores
|   |       |-- doc
|   |       |-- libraries
|   |       |-- package
|   |       |   `-- package_esp8266com_index.json
|   |       |-- platform.txt
|   |       |-- programmers.txt
|   |       |-- tests
|   |       |-- tools
|   |       |   |-- build.py
|   |       |   |-- dist
|   |       |   |   |-- esptool-0.4.5-linux32.tar.gz
|   |       |   |   |-- esptool-0.4.6-linux32.tar.gz
|   |       |   |   |-- esptool-0.4.8-linux32.tar.gz
|   |       |   |   |-- esptool-0.4.9-linux32.tar.gz
|   |       |   |   |-- linux32-xtensa-lx106-elf.tar.gz
|   |       |   |   `-- mkspiffs-0.1.2-linux32.tar.gz
|   |       |   |-- espota.py
|   |       |   |-- esptool
|   |       |   |   `-- esptool
|   |       |   |-- get.py
|   |       |   |-- get.py~
|   |       |   |-- getpy.v1
|   |       |   |-- mkspiffs
|   |       |   |   `-- mkspiffs
|   |       |   |-- sdk
|   |       |   `-- xtensa-lx106-elf
|   |       |       |-- bin
|   |       |       |   |-- xtensa-lx106-elf-addr2line
|   |       |       |   |-- xtensa-lx106-elf-ar
|   |       |       |   |-- xtensa-lx106-elf-as
|   |       |       |   |-- xtensa-lx106-elf-c++
|   |       |       |   |-- xtensa-lx106-elf-c++filt
|   |       |       |   |-- xtensa-lx106-elf-cc -> xtensa-lx106-elf-gcc
|   |       |       |   |-- xtensa-lx106-elf-cpp
|   |       |       |   |-- xtensa-lx106-elf-ct-ng.config
|   |       |       |   |-- xtensa-lx106-elf-elfedit
|   |       |       |   |-- xtensa-lx106-elf-g++
|   |       |       |   |-- xtensa-lx106-elf-gcc
|   |       |       |   |-- xtensa-lx106-elf-gcc-4.8.2
|   |       |       |   |-- xtensa-lx106-elf-gcc-ar
|   |       |       |   |-- xtensa-lx106-elf-gcc-nm
|   |       |       |   |-- xtensa-lx106-elf-gcc-ranlib
|   |       |       |   |-- xtensa-lx106-elf-gcov
|   |       |       |   |-- xtensa-lx106-elf-gdb
|   |       |       |   |-- xtensa-lx106-elf-gprof
|   |       |       |   |-- xtensa-lx106-elf-ld
|   |       |       |   |-- xtensa-lx106-elf-ld.bfd
|   |       |       |   |-- xtensa-lx106-elf-nm
|   |       |       |   |-- xtensa-lx106-elf-objcopy
|   |       |       |   |-- xtensa-lx106-elf-objdump
|   |       |       |   |-- xtensa-lx106-elf-ranlib
|   |       |       |   |-- xtensa-lx106-elf-readelf
|   |       |       |   |-- xtensa-lx106-elf-size
|   |       |       |   |-- xtensa-lx106-elf-strings
|   |       |       |   `-- xtensa-lx106-elf-strip
|   |       |       |-- build.log.bz2
|   |       |       |-- include
|   |       |       |-- lib
|   |       |       |   |-- gcc
|   |       |       |   |   `-- xtensa-lx106-elf
|   |       |       |   |       `-- 4.8.2
|   |       |       |   |-- ldscripts
|   |       |       |   `-- libiberty.a
|   |       |       |-- libexec
|   |       |       |-- share
|   |       |       |   |-- gcc-4.8.2
|   |       |       |   `-- gdb
|   |       `-- variants
|   |-- platform.keys.rewrite.txt
|   |-- platform.txt
|   `-- tools
|       `-- avr

1175 directories, 6345 files

```

devyte, 2017-11-13:
@gobo7 I'm confused. Are you trying to install the 2.3.0 stable core, or latest git core?
Although 2.3.0 is the latest stable, it is very old. I suggest trying latest git. The install instructions for installing latest git are on readthedocs. You must uninstall 2.3.0 before following the instructions (make sure the entire 2.3.0 directory is gone.

gobo7, 2017-11-13:
i am installing 2.3.0 stable and, for the time being, prefer to stick with it.

gobo7, 2017-11-15:
the git install for 2.4.0-rc2 is the same going back to 2.0.0.  so i
ran through the 2.3.0 install again with a clean arduino 1.6.9 install.

this time i found where the error is, but i don't know why or if my 
solution really works.

get.py blows up with incorrect hash for linux32-xtensa-lx106-elf.tar.gz.

package_esp8266.template.json has this sum for linux32-xtensa-lx106-elf.tar.gz:
```
f693946288f2ffa17288ef75ae16fa08573993f2b0a2a5e6bc35a68dc6087443
```
this sum is actually for the linuxarm-xtensa-lx106-elf-g46f160f-2.tar.gz.

the actual sum for the linux32 file is (downloaded from git several times):

```
b24817819f0078fb05895a640e806e0aca9aa96b47b80d2390ac8e2d9ddc955a
```

so if i edit package_esp8266.template.json and use the actual sum for
the linux32 xtensa file, get.py completes.  with this "modified" fresh
install, the blink example compiles without error.

i was asked why 2.3.0 and the answer is that i'm using arduino 1.6.9 due
to the older version of slackware.  i plan on trying 2.4.0-rc2 with
1.6.9, but will this work?  it specifies 1.8.2.

thanks.

devyte, 2018-03-02:
@gobo7 is this issue still valid with 2.4.0? with latest git

patrickelectric, 2018-06-08:
I'm getting `fork/exec /bin/xtensa-lx106-elf-g++: no such file or directory` after I updated this package to the last version.
```
arduino --pref boardsmanager.additional.urls="http://arduino.esp8266.com/stable/package_esp8266com_index.json" --save-prefs
arduino --install-boards esp8266:esp8266

```

odi, 2018-07-09:
Which OS do you use? I had the same problem with my NixOS where the interpreter-paths and the dynamic linked libraries are all in the /nix/store and this paths in the executable are wrong. Maybe the paths in your OS and the executables doesn't fit together.

I used the program [patchelf](https://github.com/NixOS/patchelf) to change the dynamic loader interpreter and the RPATH.

devyte, 2018-07-13:
Closing due to lack of feedback from original poster.

closed 2018-07-13 by devyte