esp8266/Arduino #3828: Compilation fails when using -O0

kubark42, 2017-11-13:
### Basic Infos

#### Hardware
Hardware:			Huzzah
Core Version:      	2.3.0

### Description
(Tests are done by changing all `-Os` in `platform.txt`)

Compiling fails when using -O0. It's looking for `ETS_INTR_WITHINISR`.

Compiling works with `-O1`, `-O2`, `-O3`, and `-Os`

### Settings in IDE

Module:  			Huzzah
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL

### Sketch

#### BareMinimum.ino
```cpp
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Messages

```
Arduino: 1.8.5 (Mac OS X), Board: "Adafruit HUZZAH ESP8266, 80 MHz, 115200, 4M (3M SPIFFS)"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/kenz/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/kenz/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/kenz/Documents/Arduino/libraries -fqbn=esp8266:esp8266:huzzah:CpuFrequency=80,UploadSpeed=115200,FlashSize=4M3M -ide-version=10805 -build-path /var/folders/tw/kx34lx_d52g1b6fqn483bt0w0000gn/T/arduino_build_267067 -warnings=all -build-cache /var/folders/tw/kx34lx_d52g1b6fqn483bt0w0000gn/T/arduino_cache_607096 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=/Users/kenz/Library/Arduino15/packages/esp8266/tools/esptool/0.4.9 -prefs=runtime.tools.mkspiffs.path=/Users/kenz/Library/Arduino15/packages/esp8266/tools/mkspiffs/0.1.2 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/Users/kenz/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2 -verbose /Applications/Arduino.app/Contents/Java/examples/01.Basics/BareMinimum/BareMinimum.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/kenz/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/kenz/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/kenz/Documents/Arduino/libraries -fqbn=esp8266:esp8266:huzzah:CpuFrequency=80,UploadSpeed=115200,FlashSize=4M3M -ide-version=10805 -build-path /var/folders/tw/kx34lx_d52g1b6fqn483bt0w0000gn/T/arduino_build_267067 -warnings=all -build-cache /var/folders/tw/kx34lx_d52g1b6fqn483bt0w0000gn/T/arduino_cache_607096 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=/Users/kenz/Library/Arduino15/packages/esp8266/tools/esptool/0.4.9 -prefs=runtime.tools.mkspiffs.path=/Users/kenz/Library/Arduino15/packages/esp8266/tools/mkspiffs/0.1.2 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/Users/kenz/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2 -verbose /Applications/Arduino.app/Contents/Java/examples/01.Basics/BareMinimum/BareMinimum.ino
Using board 'huzzah' from platform in folder: /Users/kenz/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0
Using core 'esp8266' from platform in folder: /Users/kenz/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0
Detecting libraries used...
"/Users/kenz/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/Users/kenz/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/include" "-I/Users/kenz/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lwip/include" "-I/var/folders/tw/kx34lx_d52g1b6fqn483bt0w0000gn/T/arduino_build_267067/core" -c -w -O0 -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10805 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP12"  -DESP8266 "-I/Users/kenz/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266" "-I/Users/kenz/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/variants/adafruit" "/var/folders/tw/kx34lx_d52g1b6fqn483bt0w0000gn/T/arduino_build_267067/sketch/BareMinimum.ino.cpp" -o "/dev/null"
Generating function prototypes...
"/Users/kenz/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/Users/kenz/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/include" "-I/Users/kenz/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lwip/include" "-I/var/folders/tw/kx34lx_d52g1b6fqn483bt0w0000gn/T/arduino_build_267067/core" -c -w -O0 -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10805 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP12"  -DESP8266 "-I/Users/kenz/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266" "-I/Users/kenz/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/variants/adafruit" "/var/folders/tw/kx34lx_d52g1b6fqn483bt0w0000gn/T/arduino_build_267067/sketch/BareMinimum.ino.cpp" -o "/var/folders/tw/kx34lx_d52g1b6fqn483bt0w0000gn/T/arduino_build_267067/preproc/ctags_target_for_gcc_minus_e.cpp"
"/Applications/Arduino.app/Contents/Java/tools-builder/ctags/5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/var/folders/tw/kx34lx_d52g1b6fqn483bt0w0000gn/T/arduino_build_267067/preproc/ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"/Users/kenz/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/Users/kenz/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/include" "-I/Users/kenz/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lwip/include" "-I/var/folders/tw/kx34lx_d52g1b6fqn483bt0w0000gn/T/arduino_build_267067/core" -c -Wall -Wextra -O0 -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10805 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP12"  -DESP8266 "-I/Users/kenz/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266" "-I/Users/kenz/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/variants/adafruit" "/var/folders/tw/kx34lx_d52g1b6fqn483bt0w0000gn/T/arduino_build_267067/sketch/BareMinimum.ino.cpp" -o "/var/folders/tw/kx34lx_d52g1b6fqn483bt0w0000gn/T/arduino_build_267067/sketch/BareMinimum.ino.cpp.o"
Compiling libraries...
Compiling core...
"/Users/kenz/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-gcc" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/Users/kenz/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/include" "-I/Users/kenz/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lwip/include" "-I/var/folders/tw/kx34lx_d52g1b6fqn483bt0w0000gn/T/arduino_build_267067/core" -c -g -x assembler-with-cpp -MMD -mlongcalls -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10805 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP12"  -DESP8266 "-I/Users/kenz/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266" "-I/Users/kenz/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/variants/adafruit" "/Users/kenz/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/cont.S" -o "/var/folders/tw/kx34lx_d52g1b6fqn483bt0w0000gn/T/arduino_build_267067/core/cont.S.o"
"/Users/kenz/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-gcc" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/Users/kenz/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/include" "-I/Users/kenz/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lwip/include" "-I/var/folders/tw/kx34lx_d52g1b6fqn483bt0w0000gn/T/arduino_build_ [... cut ...]
/Users/kenz/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/cont_util.c:58: undefined reference to `ETS_INTR_WITHINISR'
/var/folders/tw/kx34lx_d52g1b6fqn483bt0w0000gn/T/arduino_build_267067/arduino.ar(cont_util.c.o): In function `cont_can_yield':
/Users/kenz/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/cont_util.c:60: undefined reference to `ETS_INTR_WITHINISR'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Adafruit HUZZAH ESP8266.
```

justdemon, 2018-02-01:
I got the same issue, but I found the solution for it.

Using **-O0 -fkeep-inline-functions**

earlephilhower, 2019-01-22:
Closing as this is not a supported build configuration and a workaround was suggested by another user if someone really wants to hack things in their own platform.txt.

closed 2019-01-22 by earlephilhower