esp8266/Arduino #3829: Is the `division by zero` exception valid on ESP8266 or this warning should be silenced instead?

suculent, 2017-11-13:
### Basic Infos

#### Hardware
Hardware:			ESP8266
Core Version:      	2.3.0-stage

### Description

Problem description

### Settings in IDE

Module:  		Wemos D1
Flash Size: 		4M
CPU Frequency:	80Mhz
Flash Mode:		qio
Flash Frequency:	40Mhz
Upload Using:        SERIAL
Reset Method:       ck

### Sketch

Does not matter. This is a compiler warning on core.

### Debug Messages

```
In file included from /root/Arduino/hardware/esp8266com/esp8266/cores/esp8266/wiring_private.h:31:0,
                 from /root/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_pulse.c:22:
/root/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_pulse.c: In function 'pulseIn':
/root/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Arduino.h:165:44: warning: division by zero [-Wdiv-by-zero]
 #define clockCyclesToMicroseconds(a) ( (a) / clockCyclesPerMicrosecond() )
                                            ^
/root/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_pulse.c:39:37: note: in expansion of macro 'clockCyclesToMicroseconds'
     const uint32_t max_timeout_us = clockCyclesToMicroseconds(UINT_MAX);
                                     ^
/root/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Arduino.h:165:44: warning: division by zero [-Wdiv-by-zero]
 #define clockCyclesToMicroseconds(a) ( (a) / clockCyclesPerMicrosecond() )
                                            ^
/root/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_pulse.c:49:12: note: in expansion of macro 'clockCyclesToMicroseconds'
     return clockCyclesToMicroseconds(xthal_get_ccount() - pulse_start_cycle_count);
```

devyte, 2017-11-13:
@suculent you specified core 2.3.0-staged. Is this valid with latest git?
How are you building (Arduino IDE, make, pio, other)?

I don't see this warning when building normally using Arduino IDE, but then again I'm not sure which warnings are enabled there.
From what I can tell, the macro should expand correctly:

```
#define clockCyclesPerMicrosecond() ( F_CPU / 1000000L )
#define clockCyclesToMicroseconds(a) ( (a) / clockCyclesPerMicrosecond() )
```

In the Arduino IDE, I believe that the F_CPU value is specified from the menu (i.e.: boards.txt), and hence is passed as a global define via a -DF_CPU to gcc.

suculent, 2017-11-13:
It’s a headless Arduino IDE build and F_CPU is defined via —prefs build.f_cpu. 

https://github.com/suculent/arduino-docker-build

You can close this as user error, i was just posting this as general notice. 

But 0 / n is not division by zero. Is the zero in clockCyclesPerMicroseconds() possibly?

Odesláno z iPhonu

devyte, 2017-11-13:
@suculent if F_CPU is for some reason 0 in your case (e.g.: parse failure from boards.txt or something), then:

`clockCyclesPerMicrosecond()` will be 0 due to 0/n, and `clockCyclesToMicroseconds(a)` will be a/0, because:
```
#define clockCyclesPerMicrosecond() ( F_CPU / 1000000L )
#define clockCyclesToMicroseconds(a) ( (a) / clockCyclesPerMicrosecond() )
```
This could explain the warning.

You could print the value of F_CPU (and the other two defines above) at compile time with something like (untested):
#define STRINGIFY(_A_) #_A_
#pragma message "F_CPU=" STRINGIFY(F_CPU)

or just print it to Serial and look at it at runtime.

Closing for now, as this happens in the referenced Docker build system (i.e.: user error). However, I have to wonder if this is somehow related to the recent changes in boards.txt.

closed 2017-11-13 by devyte

suculent, 2017-11-13:
This is going on since updating Arduino from 1.8.3 to 1.8.5 in Docker image indeed. 

I had to add the f_cpu and flash_size params manually, otherwise those were blank. After that, the build was fixed to normal. 

However the warning remains. 

Odesláno z iPhonu