esp8266/Arduino #3831: Receiving Data over Udp stops after receiving a byte of zeros.

jentscho, 2017-11-13:
### Basic Infos

#### Hardware
Hardware:		Wemos D1 mini
Core Version:      	2.3.0

### Description

I'm currently programming an Arduino that can control Leds over Udp.
I'm writing data over udp to the arduino, my data also contains a hex color.
When sending out a byte of zeros to turn off for example green the rest of the message isn t received by the Arduino.
To test this I'm just returning the received data.
As an example:
Send to the Arduino: 53 23 FF 00 FF
Received from the Arduino: 53 23 FF

I also looked into the packets in wireshark and i also tested it with the example for the udp connection.

Isn't there a length field in the UDP Header which determines how long the Udp Packet is?

### Settings in IDE

Module:  			ESP8266
Flash Size: 		4MB
CPU Frequency:		80Mhz

### Sketch

```cpp

void setup() {
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
  }
  Udp.begin(localUdpPort);
}

void loop() {
  int packetSize = Udp.parsePacket();
  if (packetSize)
  {
    // receive incoming UDP packets
    int len = Udp.read(incomingPacket, 255);
    if (len > 0)
    {
      incomingPacket[len] = 0;
    }
    // send back a reply, to the IP address and port we got the packet from
    Udp.beginPacket(Udp.remoteIP(), Udp.remotePort());
    Udp.write(incomingPacket);
    Udp.endPacket();
  }
}

d-a-v, 2017-11-14:
try with ```Udp.write(incomingPacket, len);```

devyte, 2017-11-14:
@jentscho the previous comment is correct. Udp.write(buffer) is equivalent to Udp.write(buffer, strlen(buffer)), which means it stops at the first null byte. 
Closing as user error.

closed 2017-11-14 by devyte