esp8266/Arduino #3832: WiFi Events handler issue

Tareq-Sulaiman, 2017-11-14:
Hi everyone.
I actually have two problems with the WiFi events that are available as in the example at : https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiEvents/WiFiEvents.ino
The first problem is that the `onSoftAPModeStationConnected` handler is being called whenever anyone tries to associate with the access point even if they don't have the password, or they try to connect by accident, the esp8266 will register their device as connected for a while until the authentication fails, it will disconnect that station and call the `onSoftAPModeStationDisconnected `.
So are there any more specific connection events in AP mode (like authentication or IP address acquiring) ??
If not, is that something you are working on or considering working on??

The second problem is with the `onSoftAPModeProbeRequestReceived` , as the esp8266 is receiving multiple probe requests if a device nearby does a WiFi scan, and I didn't find anything in the 802.11 specifications about repeated probe requests.
Is the chip maybe receiving probe requests on channels other than the channel the AP is operating in?
I'm not quite sure what is going here but the number of requests picked is not consistent and I found it generally depends on the distance (the further away the station that is sending the request, the lesser they are), also the RSSI is not consistent as well, and there is a difference in the order of 30dB between the requests from the same station!!

Thank you, and your help would be appreciated.

Tareq-Sulaiman, 2017-11-15:
Anybody?

tablatronix, 2017-11-19:
I see the probes, and they have different RSSIs from both requests printed and the other thing.
What do the SDk docs say about these.

Tareq-Sulaiman, 2017-11-19:
Unfortunately, I couldn't find any further information about what's going on anywhere.
It is also worth mentioning that I tried using some WiFi sniffer sketches available online(they use promiscuous mode), and none of them had this kind of problems.
Do you have any idea about the possible cause of such an issue?

tablatronix, 2017-11-19:
I guess one could setup another esp to generate beacon frames on demand, and test it, or use wireshark and confirm there are not 2 from router. Maybe the rtos docs mention this.

d-a-v, 2017-11-19:
@Tareq-Sulaiman regarding your first question, a callback with a macaddress could be added and called by the dhcp-server (network layer, not link layer) when an IP address is offered or confirmed to a client.

devyte, 2019-10-31:
This isn't a core issue, but rather questions about behavior of SDK functionality.
@d-a-v I would close this, except for the last comment from you. What is the current status? if not available, do you intend to pursue?

devyte, 2020-02-02:
Closing due to lack of feedback.

closed 2020-02-02 by devyte