esp8266/Arduino #3833: espcomm errors after updating to macOS 10.13.1 High Sierra

TS22082, 2017-11-14:
Hardware: Macbook Pro macOS 10.13.1 High Sierra

I believe updating to High Sierra caused the problem because code that was working RIGHT before the update, do not upload anymore. I even tried uploading the sample blinky program and it wont upload to the board either. I used 2 different USB cables to make sure that wasn't the problem. 

Module:  			        Sparkfun ESP8266 Thing Dev
CPU Frequency:		80Mhz
Upload Using:		usbserial-DN031130

The errors I'm getting now:
`An error occurred while uploading the sketch
Uploading 228336 bytes from /var/folders/vm/7zhkws111z575vj68kz3wzkw0000gn/T/arduino_build_776513/blickyESP8266.ino.bin to flash at 0x00000000
warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed
warning: espcomm_send_command: wrong direction/command: 0x01 0x03, expected 0x01 0x04
error: espcomm_upload_mem failed`

igrr, 2017-11-14:
Have you tried the latest esptool-ck release (available in 2.4.0-rc2)?

TS22082, 2017-11-14:
I have not manually updated anything other than the Arduino IDE and macOS. How do I go about that? After downloading the esptool-ck zip file I have what appears to be a bunch of miscellaneous files in my downloaded folder. When I run "esptool -v" I get back command not found: esptool

TS22082, 2017-11-14:
I downloaded esptool-ck and when I try to upload the Blinky sketch to arduino I get these errors now.

`error: cannot access /dev/cu.usbserial-DN031130

error: espcomm_open failed
error: espcomm_upload_mem failed`

TS22082, 2017-11-14:
kextstat | grep FTDI gets me:

`169    0 0xffffff7f835cf000 0x7000     0x7000     com.FTDI.driver.FTDIUSBSerialDriver (2.4.2) E16D9540-AF63-36AD-94C4-B1164FBD6078 <112 55 5 4 3 1>`

TS22082, 2017-11-15:
I just did a fresh install of macOS sierra, reinstalled arduino, and downloaded the board via board manager. Im still getting the same errors as before. I also donwloaded esptool-0.4.12-osx.tar.gz and got back 

```
Thomass-MacBook-Pro:~ thomassmith$ /Users/thomassmith/Downloads/esptool-0.4.12-osx/esptool ; exit;
error: No arguments given. Use -h for help.
logout
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

[Process completed]
```

now when I run kxstat | grep FTDI I get:
'164    0 0xffffff7f85b42000 0x3000     0x3000     com.apple.driver.AppleUSBFTDI (6.0.0) 4AB9F7CD-9185-32E6-8471-053EB5E81C35 <163 120 21 5 4 3 1>'

TS22082, 2017-11-16:
Its working now! I really don't know what changed, I just slept on it and now its working again.

closed 2017-11-16 by TS22082

DIYglenn, 2017-11-17:
I'm still having some issues with this. I get the exact same errors:
`warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed
warning: espcomm_send_command: wrong direction/command: 0x01 0x03, expected 0x01 0x04
error: espcomm_upload_mem failed`

I'm trying to update my "Digistump Oak's", and after power cycling and retrying, same error.

I tried again on my Mac Pro with El Capitan - No issues with Arduino 1.8.5!
Something is broken with High Sierra...

AhmedAKhan, 2017-11-18:
I am getting the original error as the Original post, what have you tried before it started working?

TS22082, 2017-11-18:
I downloaded:
https://github.com/igrr/esptool-ck/releases/tag/0.4.12

and the FTDI driver for my system from:
http://www.ftdichip.com/Drivers/VCP.htm

Shutdown my computer and came back to it the next day and it was working.

johnsonberry, 2018-09-19:
I also installed of macOS sierra, and then reinstalled Arduino, and downloaded the board via the board manager. [Brother Printer keeps going offline Error](https://www.brotherprintersupportnumber.com/blog/how-to-fix-printer-keeps-going-offline-error/)