esp8266/Arduino #3837: Git 2.4.0 rc2 version wifi client example lwip2 crash

tontito, 2017-11-15:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			ESP-07
Core Version:      	2.4.0-rc2 git version

### Description

Problem description

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		512KB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

/*
 *  This sketch sends data via HTTP GET requests to data.sparkfun.com service.
 *
 *  You need to get streamId and privateKey at data.sparkfun.com and paste them
 *  below. Or just customize this script to talk to other HTTP servers.
 *
 */

#include <ESP8266WiFi.h>

const char* ssid     = "your-ssid";
const char* password = "your-password";

const char* host = "data.sparkfun.com";
const char* streamId   = "....................";
const char* privateKey = "....................";

void setup() {
  Serial.begin(115200);
  delay(10);

  // We start by connecting to a WiFi network

  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  
  /* Explicitly set the ESP8266 to be a WiFi-client, otherwise, it by default,
     would try to act as both a client and an access-point and could cause
     network-issues with your other WiFi-devices on your WiFi-network. */
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

int value = 0;

void loop() {
  delay(5000);
  ++value;

  Serial.print("connecting to ");
  Serial.println(host);
  
  // Use WiFiClient class to create TCP connections
  WiFiClient client;
  const int httpPort = 80;
  if (!client.connect(host, httpPort)) {
    Serial.println("connection failed");
    return;
  }
  
  // We now create a URI for the request
  String url = "/input/";
  url += streamId;
  url += "?private_key=";
  url += privateKey;
  url += "&value=";
  url += value;
  
  Serial.print("Requesting URL: ");
  Serial.println(url);
  
  // This will send the request to the server
  client.print(String("GET ") + url + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" + 
               "Connection: close\r\n\r\n");
  unsigned long timeout = millis();
  while (client.available() == 0) {
    if (millis() - timeout > 5000) {
      Serial.println(">>> Client Timeout !");
      client.stop();
      return;
    }
  }
  
  // Read all the lines of the reply from server and print them to Serial
  while(client.available()){
    String line = client.readStringUntil('\r');
    Serial.print(line);
  }
  
  Serial.println();
  Serial.println("closing connection");
}

Exception (0):
epc1=0x70bce434 epc2=0x00000000 epc3=0x4020e29b excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffc20 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffdc0:  0000015e 00000000 3fff4854 401043cd  
3ffffdd0:  3fff40c4 3fff4854 0000007f 00000000  
3ffffde0:  00000000 00000020 40100ec2 3fff3f2d  
3ffffdf0:  3fff3ef8 00000000 3fff40c4 4010451c  
3ffffe00:  3fff3ef8 3fff40c4 3fff4106 4021ea04  
3ffffe10:  00000008 3ffea3e3 3ffeebb0 40225408  
3ffffe20:  40103926 401034c3 3ffeebb0 3fff4114  
3ffffe30:  3fff40c4 3ffea4c4 3fff3ef8 40224a85  
3ffffe40:  40100000 0000007f 7fffffff 00000000  
3ffffe50:  00000014 3ffea4c4 3fff40c4 40225408  
3ffffe60:  4010228b 00080000 00000000 4000050c  
3ffffe70:  3ffea4c8 000000ff 00000000 00000011  
3ffffe80:  3fff3eac 40101f88 3fffc200 00000022  
3ffffe90:  00000011 00000001 0000001e 3fff40c4  
3ffffea0:  3ffea4c4 3fff4128 3fff5fe4 40225496  
3ffffeb0:  3fff3ef8 00000000 00000000 40225b97  
3ffffec0:  ffffffff 00000000 3ffeab81 40222077  
3ffffed0:  3fff3ef8 3fff4ab4 3ffeedfc 00000001  
3ffffee0:  3fff40c4 3fff3ef8 3ffea4c8 00000043  
3ffffef0:  3fff3ef8 00000001 3fff5f9c 40223276  
3fffff00:  3fff2aac 00000007 00000007 3fff3c58  
3fffff10:  00000000 3fff3ef8 3fff5f9c 402237b4  
3fffff20:  00000000 00000000 3fff3ef8 402242fa  
3fffff30:  4021fa44 3fff3c5c 3ffea464 4021fa52  
3fffff40:  40104dae 00b32cb0 00000000 4021fb10  
3fffff50:  3ffea464 3ffef4c0 00b32cb0 00000000  
3fffff60:  40245a58 3ffefc88 3ffefcb0 60000600  
3fffff70:  023d62bf 3ffefcb0 3ffefc88 40245a65  
3fffff80:  40245aaa 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffefcc8 3fffdad0 3fff2b90 40217737  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<

Exception 0: Illegal instruction
Decoding 31 results
0x4020e29b: loop at C:\Users\VOLKI\Desktop\Esp8266\Kombi_Ekran/Kombi_Ekran.ino line 272
0x401043cd: glue2esp_linkoutput at /home/david/dev/esp8266/lwip2v3/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c line 263
0x40100ec2: pp_post at ?? line ?
0x4010451c: new_linkoutput at /home/david/dev/esp8266/lwip2v3/tools/sdk/lwip2/builder/glue-lwip/lwip-git.c line 205
0x4021ea04: ethernet_output at /home/david/dev/esp8266/lwip2v3/tools/sdk/lwip2/builder/lwip2-src/src/netif/ethernet.c line 305
0x40225408: ip4_output_if_opt_src at /home/david/dev/esp8266/lwip2v3/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c line 962
0x40103926: lmacRecycleMPDU at ?? line ?
0x401034c3: lmacRecycleMPDU at ?? line ?
0x40224a85: etharp_output_LWIP2 at /home/david/dev/esp8266/lwip2v3/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/etharp.c line 893
0x40100000: _stext at ?? line ?
0x40225408: ip4_output_if_opt_src at /home/david/dev/esp8266/lwip2v3/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c line 962
0x4010228b: wDev_ProcessFiq at ?? line ?
0x40101f88: wDev_ProcessFiq at ?? line ?
0x40225496: ip4_output_if_src at /home/david/dev/esp8266/lwip2v3/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c line 801
0x40225b97: ip_chksum_pseudo at /home/david/dev/esp8266/lwip2v3/tools/sdk/lwip2/builder/lwip2-src/src/core/inet_chksum.c line 542
0x40222077: udp_sendto_if_src at /home/david/dev/esp8266/lwip2v3/tools/sdk/lwip2/builder/lwip2-src/src/core/udp.c line 463
0x40223276: dhcp_create_msg at /home/david/dev/esp8266/lwip2v3/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/dhcp.c line 836
0x402237b4: dhcp_discover at /home/david/dev/esp8266/lwip2v3/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/dhcp.c line 836
0x402242fa: dhcp_fine_tmr at /home/david/dev/esp8266/lwip2v3/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/dhcp.c line 470
0x4021fa44: cyclic_timer at /home/david/dev/esp8266/lwip2v3/tools/sdk/lwip2/builder/lwip2-src/src/core/timeouts.c line 165
0x4021fa52: cyclic_timer at /home/david/dev/esp8266/lwip2v3/tools/sdk/lwip2/builder/lwip2-src/src/core/timeouts.c line 171
0x40104dae: wdt_feed at ?? line ?
0x4021fb10: sys_check_timeouts at /home/david/dev/esp8266/lwip2v3/tools/sdk/lwip2/builder/lwip2-src/src/core/timeouts.c line 351
0x40245a58: ets_timer_handler_isr at ?? line ?
0x40245a65: ets_timer_handler_isr at ?? line ?
0x40245aaa: ets_timer_handler_isr at ?? line ?
0x40217737: loop_task at C:\Arduino\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_main.cpp line 57

old version 2.3.0 no problem

d-a-v, 2017-11-15:
I assume you use master (2.4.0rc2 does not have lwip2).
...
Thanks, I could reproduce your bug with a big public html page.

d-a-v, 2017-11-15:
I however have the same problem with lwip1.4 (in master).
Decoding my stack strace with lwip2 shows that the problem happens in ```String::changeBuffer()```.
So in my case it is a ```malloc()``` problem.
My page is too big for your ```client.readStringUntil('\r');```
It could be the same for you too, happening in some other place (```malloc()``` is used everywhere).
The current SDK is known to eat more RAM than the one used in 2.3.0. Your data may filling to the edge.
If it really is a RAM problem, you should read your page chunk by chunk, not all at once.
How big is your webpage ?
Can you print ```ESP.getFreeHeap()``` after reading your data with 2.3.0 ?

tontito, 2017-11-15:
this error no ram problem. 
if rssi -82 -85 no problem if rssi -92 crash

d-a-v, 2017-11-15:
Can you try it with lwip1.4 (select it in ```menu>Tools>LwIP Variant>Prebuilt Source (gcc)```) ?

tontito, 2017-11-15:
Prebuilt Source (gcc) no problem
Espressif xcc no problem
v2 problem but only -92 -95 db rssi

d-a-v, 2017-11-15:
Can you print ```ESP.getFreeHeap()``` after reading your data ?
What is your data size once loaded ?
(poor link, fragmented data, more malloc())

tontito, 2017-11-15:
Prebuilt Source (gcc) no problem lwip 1.4 no problem 3 hours.

tomorrow test lwip2

tontito, 2017-11-28:
I couldn't find problem. Prebuilt Source (gcc) no problem lwip 1.4

closed 2017-11-28 by tontito

devyte, 2017-11-29:
@tontito I believe the last question was whether it works with lwip 1.4, and not with lwip 2. In that case, the issue should be investigated, because we're trying to migrate

schlaegerz, 2017-12-24:
Just to add to this. I was playing with it and got a very similar crash on lwip 2. I have never seen it on 1.4.