esp8266/Arduino #3842: 2.4-rc2 crash while connecting to Eduroam

frischj, 2017-11-17:
Hi,

I try to connect the ESP8266 with eduroam with the new 2.4-rc2 Arduino lib pulled via git as in the descriptions. The code connects to Eduroam as I can see connections results in our university log (only one line, sorry no output of log from radius). At some point, I get a crash and an error stack. I paste the code and the error stack below.

#include <ESP8266WiFi.h>

extern "C" {
  #include "user_interface.h"
  #include "wpa2_enterprise.h"
}

// SSID to connect to
char ssid[] = "eduroam";
char wlpasswd[] = "";
char username[] = "--------------------";
char identity[] = "------------------";
char password[] = "-------------";
char new_password[] = "-----------";

const char* host = "ipinfo.io";

void setup() {
  Serial.begin(115200);
  delay(1000);
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.mode(WIFI_STA);
  
  wifi_station_set_wpa2_enterprise_auth(1);

  // Clean up to be sure no old data is still inside
  wifi_station_clear_cert_key();
  wifi_station_clear_enterprise_ca_cert();
  wifi_station_clear_enterprise_identity();
  wifi_station_clear_enterprise_username();
  wifi_station_clear_enterprise_password();
  wifi_station_clear_enterprise_new_password();
    
  wifi_station_set_enterprise_identity((uint8*)identity, strlen(identity));
  wifi_station_set_enterprise_username((uint8*)username, strlen(username));
  wifi_station_set_enterprise_password((uint8*)password, strlen(password));
  wifi_station_set_enterprise_new_password((uint8*)new_password, strlen(new_password));
  //wifi_station_set_enterprise_ca_cert(ca_pem, ca_pem_len);
  
  WiFi.begin(ssid, wlpasswd);
    
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.print(".");
  }

  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}
 
void loop() {
  delay(5000);
 
  Serial.print("connecting to ");
  Serial.println(host);
  
  WiFiClient client;
  const int httpPort = 80;
  if (!client.connect(host, httpPort)) {
    Serial.println("connection failed");
    return;
  }
  
  client.print(String("GET ") + "/ip" + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" + 
               "Connection: close\r\n\r\n");
  delay(500);
  
  while(client.available()){
    String line = client.readStringUntil('\r');
    Serial.print(line);
  }
  return;
}

The console result reads:

load 0x4010f000, len 1384, room 16 

tail 8

chksum 0x2d

csum 0

Connecting to eduroam

....

Exception (28):

epc1=0x4000df60 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffd70 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff10:  402187f9 3ffebd29 00000000 00000000  

3fffff20:  40223496 0ee83a2c 2300baa7 d02739eb  

3fffff30:  40225346 5060d027 3fff0007 00000000  

3fffff40:  3fff211c 4021cd6f 3ffee130 3ffec92c  

3fffff50:  3ffec92c 0000002e 00000000 00000035  

3fffff60:  00000002 0000001a 40230caf 3ffee130  

3fffff70:  3ffec920 3fffdcc0 3ffea7c0 3ffea7c0  

3fffff80:  00000080 3ffee130 00000000 3fffdcc0  

3fffff90:  4023058b 3fffdab0 00000000 402035c3  

3fffffa0:  3ffea7c0 40000f49 3fffdab0 40000f49  

<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

And the decoded error stack reads:

Decoding 9 results
0x402187f9: ieee80211_decap at ?? line ?
0x40223496: wpa_set_pmk at ?? line ?
0x40225346: wpa2_sm_rx_eapol at ?? line ?
0x4021cd6f: sta_input at ?? line ?
0x40230caf: pp_tx_idle_timeout at ?? line ?
0x4023058b: ppPeocessRxPktHdr at ?? line ?
0x402035c3: loop_task at /opt/arduino-1.8.5/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 57

Anybody some ideas, if I make something wrong?

martinius96, 2017-12-29:
Can you please tell me if you solved it?

L0laapk3, 2018-01-29:
I can confirm that this code works.

For future reference to people finding this from google: use `username` = `identity` = _name@university.domain_ and `password` = `new_password` =_your password_

devyte, 2018-01-30:
@L0laapk3 do you mean that the sketch in the original post works for you?

martinius96, 2018-01-30:
send us sketch please.

L0laapk3, 2018-01-31:
The sketch is literally in the first post lol

lzydavid, 2018-04-20:
hi, my version is sdk 2.2.1/ core 2.4.1 , i tried connecting to my school network but still get errors: 

SDK:2.2.1(cfd48f3)/Core:2.4.1/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1)
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)

Connecting to NTUSECURE
WPA2 ENTERPRISE VERSION: [v2.0] enable
scandone
state: 3 -> 2 (b0)
.state: 2 -> 0 (2)
reconnect
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 
EAP-PEAP: received 37 bytes encrypted data for Phase 2
EAP-PEAP: received Phase 2: code=1 identifier=50 length=5
EAP-PEAP: Phase 2 Request: type=1
.EAP-PEAP: received 69 bytes encrypted data for Phase 2
EAP-PEAP: received Phase 2: code=1 identifier=51 length=35
EAP-PEAP: Phase 2 Request: type=26
EAP-PEAP: Selected Phase 2 EAP vendor 0 method 26
EAP-MSCHAPV2: RX identifier 51 mschapv2_id 51
EAP-MSCHAPV2: Generate Challenge Response
.EAP-PEAP: received 85 bytes encrypted data for Phase 2
EAP-PEAP: received Phase 2: code=1 identifier=52 length=51
EAP-PEAP: Phase 2 Request: type=26
EAP-MSCHAPV2: RX identifier 52 mschapv2_id 51
.EAP-PEAP: received 37 bytes encrypted data for Phase 2
EAP-PEAP: received Phase 2: code=3 identifier=52 length=4
EAP-PEAP: Phase 2 Success
.Fatal exception 3(LoadStoreErrorCause):
epc1=0x40100326, epc2=0x00000000, epc3=0x00000000, excvaddr=0x40243684, depc=0x00000000

Exception (3):
epc1=0x40100326 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40243684 depc=0x00000000

ctx: sys 
sp: 3ffffc80 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffe20:  40101efd 3ffedd58 3ffee4c0 00000001  
3ffffe30:  00007fff 0029c2eb 3ffee7dc 401020ce  
3ffffe40:  3ffefe6c 00000244 00000244 4010020c  
3ffffe50:  40245460 00000000 3fff26bc 401009a8  
3ffffe60:  7fffffff 3ffeabfc 3fff26c4 40107198  
3ffffe70:  3ffefe6c 00000450 00000450 4021197e  
3ffffe80:  3ffefe6c 0000029d 0000029d 4010020c  
3ffffe90:  3ffefe6c 00000270 00000270 3ffeef78  
3ffffea0:  00000008 3ffee71c 3fff22a8 401004fe  
3ffffeb0:  3ffefe6c 0000017e 0000017e 4010020c  
3ffffec0:  3ffefe6c 0000017b 0000017b 4010020c  
3ffffed0:  40245460 3fff1d9c 3fff2074 401009a8  
3ffffee0:  3ffefe6c 00000150 00000150 4010020c  
3ffffef0:  00000008 3ffee71c 3fff22ac 4010053a  
3fffff00:  40226779 3fff1f54 3fff1d9c 4010099c  
3fffff10:  402254b0 3fff1f54 3fff1d9c 40107198  
3fffff20:  40224d7a 3fff1d9c 3ffee71c 40224d8e  
3fffff30:  40224dc9 3fff1d9c 00000000 40224994  
3fffff40:  4021c33d 3ffedd58 3ffedd58 3ffecb70  
3fffff50:  3ffecb70 00000054 00000000 00000029  
3fffff60:  00000002 0000001a 40230303 3ffedd58  
3fffff70:  3ffecb64 3fffdcc0 3ffea3a0 3ffea3a0  
3fffff80:  00000080 3ffedd58 00000000 3fffdcc0  
3fffff90:  4022fbc3 3fffdab0 00000000 00000001  
3fffffa0:  3ffea3a0 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld

Anyone got any idea how to fix this?

d-a-v, 2019-06-23:
[https://github.com/espressif/ESP8266_RTOS_SDK/issues/635](http://comment/if/you/can)