esp8266/Arduino #3843: website that serves Json file down?

alexmac131, 2017-11-17:
I get this message when attempting to connect after not being able to connect through the arduino board manager..   The directions in this git point specifically  to - its down at time of this writing
"http://arduino.esp8266.com/stable/package_esp8266com_index.json"   

This site can’t be reached

arduino.esp8266.com refused to connect.
Try:
Checking the connection
Checking the proxy and the firewall
ERR_CONNECTION_REFUSED

absk1s3, 2017-11-17:
I have the same issue, it seems like arduino.esp8266.com is down.

FsFalcon, 2017-11-17:
seeing the same

maslaff, 2017-11-17:
Подтверждаю - упало...

kegrad, 2017-11-17:
This was also discussed on the arduino forums: https://forum.arduino.cc/index.php?topic=461880.0 
It suggests the alternative address https://github.com/esp8266/Arduino/releases/download/2.3.0/package_esp8266com_index.json which seems to work for me.

arthur12320, 2017-11-17:
@kegrad changing to that link also worked for me. thanks for the help!

rjp663, 2017-11-18:
Thank you!!! The alternate URL worked for me.

closed 2017-11-21 by igrr