esp8266/Arduino #3844: Esp Serial write not available

bigelite, 2017-11-18:
hi all I've been working on semi IOT project that is simple dimmer with lamp lighting control, when i want to set value that receive from android app to serial, its not work correctly. all the hardware are good set because for example this code is working:

`int i =0;
void setup() {
  pinMode(2, OUTPUT);
  Serial.begin(9600);
}
void loop()
{
  digitalWrite(2,LOW);
  delay(100);
  digitalWrite(2,HIGH);
  delay(100);
    for (i =1 ; i<9 ; i+=1){
      if(Serial.available())      Serial.write(i); else     
 Serial.println("esp:serial w not available");
      Serial.println(i);
      delay(3000);
    }
    for (i =9 ; i>0 ; i-=1){
      if(Serial.available())      Serial.write(i); else     
      Serial.println("esp:serial w not available");
      Serial.println(i);
      delay(3000);
    }  
}`

this code work correctly and dimming lamp light increase and decrease, but when i want to receive value from android like this it doesn't work:

` #include <ESP8266WiFi.h>

IPAddress staticIP832_240(192, 168, 43, 240);
IPAddress gateway832_240(192, 168, 43, 1);
IPAddress subnet832_240(255, 255, 255, 0);
const char *ssid = "G3";
const char *password = "@4112303";
WiFiServer server(80);
int value = 0;
void setup() {
    Serial.begin(9600);
    pinMode(2, OUTPUT);

    delay(3000);
    WiFi.disconnect();
    Serial.println("Start");
    WiFi.begin(ssid, password);
    while ((!(WiFi.status() == WL_CONNECTED))) {
        delay(300);
        Serial.print("...");
    }
    Serial.println("");
    Serial.println("WiFi connected");

    WiFi.config(staticIP832_240, gateway832_240, subnet832_240);
    WiFi.hostname("G3_ESP");
    server.begin();
    Serial.println("Server started");
    Serial.print("Use this URL to connect: ");
    Serial.print("http://");
    Serial.print(WiFi.localIP());
    Serial.println("/");

    delay(1000);
    Serial.println("");
    Serial.println("Your IP Addressees:  ");
    Serial.print((WiFi.localIP()));
    Serial.println("");
    Serial.println("Your Gate Way:  ");
    Serial.print((WiFi.gatewayIP().toString().c_str()));
    Serial.println("");
    Serial.println("Your Host Name:  ");
    Serial.print((WiFi.hostname()));
}

void loop() {
        // Check if a client has connected
        WiFiClient client = server.available();
        if (!client) {
            return;
        }
        // Wait until the client sends some data
        Serial.println("new client");
        while (!client.available()) {
            delay(1);
        }
        // Read the first line of the request
        String request = client.readStringUntil('\r');
        Serial.println(request);
        client.flush();
        // Match the request
        if (Serial.available()) {
            Serial.write(2);
            Serial.write(8);
            Serial.write(0);
        } else {
            client.println("serial not available in first show up");
        }
        if (request.indexOf("/up") != -1) {
            if (Serial.available()) {
                Serial.write(8);
                client.println("up is successful");
            } else {
                client.println("up not successful");
            }
            return;
        } else if (request.indexOf("/down") != -1) {
            if (Serial.available()) {
                Serial.write(2);
                client.println("down is successful");
            } else {
                client.println("down not success");
            }
            return;
        }
        client.println("HTTP/1.1 200 OK");
        client.println("Content-Type: text/html");
        client.println(""); //  do not forget this one
        client.println("<!DOCTYPE HTML>");
        client.println("<html>");
        client.print("lamp value is : ");
        client.print(value);
        client.println("<br><br>");
        client.println("<a href=\"/up\"\"><button>up</button></a>");
        client.println("<a href=\"/down\"\"><button>down</button></a><br />");
        client.println("</html>");

        delay(1);
        Serial.println("Client disonnected");
        Serial.println("");
        client.println("exception has occurred ");
        return;

    }

}`

igrr, 2018-01-02:
> but when i want to receive value from android like this it doesn't work

Could you please elaborate, what exactly does not work? Please try to narrow down the issue you are seeing.

devyte, 2018-02-08:
Closing due to lack of feedback in over a month.

closed 2018-02-08 by devyte