esp8266/Arduino #3850: Upload speed ignored

AndreKR, 2017-11-19:
#### Hardware
Hardware: ESP-12F

While debugging #3481 I was switching between 2.3.0 and the latest master and I noticed, that with the latest Git master, the upload is much slower:

2.3.0: "Uploading 232352 bytes ..." -> 11 seconds
master: "Uploading 257792 bytes ..." -> 25 seconds
Upload speed is set to 256000 in both cases.

Then I set the upload speed to 115200 and found that it still takes 25 seconds to upload. So the upload speed seems to be ignored in the current Git master.

When I check the command line of esptool in Process Explorer, I find:
```
C:\Users\andre\Documents\Arduino\hardware\esp8266com\esp8266\tools\esptool\esptool.exe -cd ck -cb 115200 -cp COM22 -ca 0x00000 -cf C:\Users\andre\AppData\Local\Temp\arduino_build_268796/udptest.ino.bin
```
This is with the upload speed set to "256000".

d-a-v, 2017-11-19:
This is related with #3722, I will have a look to this.

d-a-v, 2017-11-19:
@AndreKR Can you try this [boards.txt](https://gist.githubusercontent.com/d-a-v/8b19d5dff5608d737146fcecb118e24d/raw/daaf7286172152a56e8bc5dc307d59e89d10b073/boards.txt) and check is serial upload speed is restored ?

PR #3940 (2017-12-12): update and fix current generated boards.txt
fix 16M spiffs
less confusing lwip versions in menus
fix #3850
fix #3707
fix #3640
fix #3098
fix #2740
fix #2683
fix #2550
commit bd261c05 (2017-12-12): update and fix current generated boards.txt (#3940)

less confusing lwip versions in menus
fix #3850
fix #3707
fix #3640
fix #3098
fix #2740
fix #2683
fix #2550
closed 2017-12-12 by devyte