esp8266/Arduino #3852: wdt reset in WiFi.Config(...)

xbary, 2017-11-19:
### Basic Infos

#### Hardware
Hardware:		?ESP-07S?
Core Version:      	?2.4.0-rc2?

### Description

The problem is that after losing the wifi network.
When trying again to connect to wifi, pops up a watch dog reset.

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:	?160Mhz?
Flash Mode:		?dio?
Flash Frequency:	?40Mhz?
Upload Using:		?SERIAL?
Reset Method:		?nodemcu?
IwIPvariant:             ?v2?

### Sketch

```cpp
#include <WiFiUdp.h>
#include <WiFiServer.h>
#include <WiFiClientSecure.h>
#include <WiFiClient.h>
#include <ESP8266WiFiType.h>
#include <ESP8266WiFiSTA.h>
#include <ESP8266WiFiScan.h>
#include <ESP8266WiFiMulti.h>
#include <ESP8266WiFiGeneric.h>
#include <ESP8266WiFiAP.h>
#include <ESP8266WiFi.h>

IPAddress ip(192, 168, 1, 32);
IPAddress gateway(192, 168, 1, 1);
IPAddress mask(255, 255, 255, 0);

void setup()
{
	Serial.begin(115200);
	Serial.setDebugOutput(true);

	WiFi.persistent(false);
	WiFi.setAutoConnect(false);
	WiFi.setAutoReconnect(false);
	WiFi.mode(WIFI_STA);
	WiFi.disconnect();
	delay(100);
	Serial.println("\n\nStart\n");
}

void loop()
{
	static uint32_t tick = millis();
	static bool ScanNetworks = false;
	static bool WiFiFind = false;
	static bool WiFiAfterInit = false;

	if (millis() - tick > 1000)
	{
		tick = millis();
		Serial.printf("\ntick:%d / freeheap:%d / status:%d\n", (tick / 1000), ESP.getFreeHeap(), WiFi.status());

		if ((WiFi.status() == WL_IDLE_STATUS) || ((WiFi.status() == WL_DISCONNECTED) && (WiFiAfterInit==false)))
		{
			if (WiFiFind)
			{
				
				WiFi.mode(WIFI_STA);
				WiFi.hostname("xb_IOTTEST");

				Serial.print("\nWiFi.config(ip, gateway, mask);");
				WiFi.config(ip, gateway, mask);
				Serial.print("...OK");

				Serial.print("\nWiFi.begin(\"xb_HUAWEI\", \"0987654321\");");
				WiFi.begin("xb_HUAWEI", "0987654321");
				Serial.print("...OK");

				WiFiFind = false;
				WiFiAfterInit = true;
			}
			else if (!ScanNetworks)
			{
				WiFiAfterInit = false;
				int n = WiFi.scanNetworks(true);
				if (n == 0)
				{
					Serial.println("\n\nNo network found.\n");
				}
				else
				{
					Serial.println("\n\nSTART scan networks.\n");
					ScanNetworks = true;
				}
			}
			else
			{
				int n = WiFi.scanComplete();
				if (n == WIFI_SCAN_RUNNING)
				{
					Serial.println("\n\nBusy scan networks.\n");
				}
				else if (n == WIFI_SCAN_FAILED)
				{
					Serial.println("\n\nFailed scan networks.\n");
					ScanNetworks = false;
				}
				else if (n == 0)
				{
					Serial.println("\n\nNo network found.\n");
					ScanNetworks = false;

				}
				else
				{
					WiFiFind = false;
					for (int i = 0; i < n; ++i)
					{
						// Print SSID and RSSI for each network found
						Serial.print(i + 1);
						Serial.print(": ");
						Serial.print(WiFi.SSID(i));
						if (WiFi.SSID(i) == "xb_HUAWEI")
						{
							WiFiFind = true;
						}

						Serial.print(" (");
						Serial.print(WiFi.RSSI(i));
						Serial.print(")");
						Serial.println((WiFi.encryptionType(i) == ENC_TYPE_NONE) ? " " : "*");
						delay(50);
					}
					WiFi.scanDelete();
					ScanNetworks = false;
				}
			}
		}
	}
}

```

### Debug Messages

```
Opening port
Port open

tick:28 / freeheap:46120 / status:0
1: TP-LINK (-82)*
2: xb_HUAWEI (-47)*
3: xb_modecom (-70)*

tick:29 / freeheap:46328 / status:0

WiFi.config(ip, gateway, mask);STUB: dhcp_stop
...OK
WiFi.begin("xb_HUAWEI", "0987654321");...OK
tick:30 / freeheap:46352 / status:6

tick:31 / freeheap:46352 / status:6
scandone
state: 0 -> 2 (b0)

tick:32 / freeheap:46352 / status:6
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 7
cnt 

connected with xb_HUAWEI, channel 11
ip:192.168.1.32,mask:255.255.255.0,gw:192.168.1.1
ip:192.168.1.32,mask:255.255.255.0,gw:192.168.1.1

tick:33 / freeheap:45712 / status:3

tick:34 / freeheap:45712 / status:3

....

tick:40 / freeheap:45712 / status:3

tick:41 / freeheap:45712 / status:3
pm open,type:2 0

tick:42 / freeheap:45208 / status:3
.....
tick:124 / freeheap:45208 / status:3

tick:125 / freeheap:45208 / status:3

tick:126 / freeheap:45208 / status:3

tick:127 / freeheap:45208 / status:3     <------- This moment is router restart
bcn_timout,ap_probe_send_start

tick:128 / freeheap:45208 / status:3

tick:129 / freeheap:45208 / status:3
ap_probe_send over, rest wifi status to disassoc
state: 5 -> 0 (1)
rm 0
pm close 7

tick:130 / freeheap:46048 / status:0

START scan networks.

tick:131 / freeheap:45976 / status:0

Busy scan networks.

tick:132 / freeheap:45976 / status:0

Busy scan networks.

tick:133 / freeheap:45904 / status:0

Busy scan networks.

tick:134 / freeheap:45904 / status:0

Busy scan networks.

tick:135 / freeheap:45904 / status:0

Busy scan networks.

scandone

tick:136 / freeheap:45904 / status:0
1: xb_modecom (-72)*
2: TP-LINK (-79)*

tick:137 / freeheap:46048 / status:0

............

START scan networks.

tick:166 / freeheap:46048 / status:0

Busy scan networks.

tick:167 / freeheap:45976 / status:0

Busy scan networks.

tick:168 / freeheap:45904 / status:0

Busy scan networks.

tick:169 / freeheap:45832 / status:0

Busy scan networks.

tick:170 / freeheap:45832 / status:0

Busy scan networks.

scandone

tick:171 / freeheap:45840 / status:0
1: xb_modecom (-72)*
2: TP-LINK (-80)*
3: xb_HUAWEI (-49)*

tick:172 / freeheap:46048 / status:0

WiFi.config(ip, gateway, mask);
 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```

d-a-v, 2017-11-19:
This should be fixed in [lwip2](https://github.com/d-a-v/esp82xx-nonos-linklayer/issues/1) via #3835.
Could you confirm the fix with replacing your tools/sdk/lib/liblwip2.a by this [one](https://github.com/d-a-v/Arduino/raw/214f0001e2817065a435356861aabab046bb1383/tools/sdk/lib/liblwip2.a) ?

xbary, 2017-11-19:
bingo, it works!

I had to change the library, of course, but I also added this function as:
file time.c add:
```
int settimeofday(const struct timeval* tv, const struct timezone* tz)
{
	if (tz) /*before*/
	{
	//	sntp_set_timezone(((int)tz->tz_minuteswest) / 60);
		// apparently tz->tz_dsttime is a bitfield and should not be further used (cf man)
	//	sntp_set_daylight(0);
	}
	if (tv) /* after*/
	{
	//	sntp_set_system_time(tv->tv_sec);
		// ignore tv->usec
	}
	return 0;
}
```

And now, to be able to consolidate the code I had to comment as above.

The compilation ran without error, the re-connection test after losing the network came out positively. I applied my code shown above without modification.

Thank you
ps. I hope now that the official github will appear with the patch.

d-a-v, 2017-11-19:
Thanks for testing :)
@igrr should I squash #3835 ?

commit cbfbc1ad (2017-11-21): lwip2 fixes and time/ntp management
core: +settimeofday()
core: +coredecls.h +sntp-lwip2.c
core: fix clock_gettime() with micros64()
core: honor DST in configTime()
core: internal clock is automatically started
examples: +esp8266/NTP-TZ-DST.ino
lwip2: sntp client removed
lwip2: fix crashing with WiFi.softAPConfig(ip,ip,ip)
fix #3852
closed 2017-11-21 by igrr