esp8266/Arduino #3853: Travis CI is hitting the 50min limit too often

devyte, 2017-11-20:
### Basic Infos

#### Hardware
Core Version:      	latest git

### Description

Travis CI is hitting the 50min limit more often than not, even when only a single build is running. As a result, builds need to be restarted often, sometimes several times in a row, before it runs to completion.

devyte, 2017-11-20:
@igrr I'm currently stuck, because travis CI keeps failing due to the timelimit. It has consistently failed several times in a row now.
I know exactly nothing about travis ci, and after an hour or so of googling, I still don't really have a concrete clue as to how to proceed. 
Help?

igrr, 2017-11-20:
After a brief look, i can't find why it started failing all of a sudden.
Will work on splitting the build into separate jobs (at least two: one for Arduino IDE builds, one for platformio).

devyte, 2017-11-20:
@igrr From what I can see, a normal build run is taking 42-45 minutes. I suppose it's possible there is too much load on the host cluster these days?
From my googling I gathered that the build runs on a 2-core system. Does make run with -j 2? If not, that might give another speed boost.

devyte, 2017-11-20:
Not the right place to ask this, but what do I do with issues after I have merged a fix? Do I just add label:staged-for-release, or is there something else I should do? Like I said elsewhere, I'm looking at issues with label:easy that are marked with milestone 2.4.

igrr, 2017-11-20:
When building using Arduino IDE, `make` is not used, so we can't easily enable parallel builds there. Platformio, as far as i know, does parallel builds by default.
Splitting the whole build into multiple jobs looks like the easiest option.

Regarding tagging, for issues which have been present in previous release (or pre-release), i suggest tagging with "staged-for-release" and keeping open, so that the issue still pops up in search in case someone else encounters it. For regressions introduced in master after the last release (or pre-release), it is okay to just close the issue.

devyte, 2017-12-08:
@igrr this is happening frequently again...