esp8266/Arduino #3859: Update failed!

nedimhx, 2017-11-22:
Hello, 
http update with .ino file on browser works perfectly fine on NodeMCU (ESP8266-12E) but on Generic ESP8266-12E chip everytime I got "Update failed"

Could it be due to wrong wiring ?

devyte, 2017-11-22:
Most likely. Check connections, and make sure of the state of the three relevant gpio on boot (don't remember which off the top of my head). Just google esp gpio boot.
Closing as user error.

closed 2017-11-22 by devyte

nedimhx, 2017-11-23:
@devyte but as notemcu it needs all available pwd ports for LED strip, so when I need to change working mode of led strips, I just upload .ino file, like I said on nodeMCU it works perfectly.

Regarding GPIO boot and flashing via uart it works but flashing via http not work

devyte, 2017-11-23:
The nodemcu has specific wiring of the esp12. The fact that it works on the nodemcu and not on your generic module points to a wiring issue with your generic module, or some wrong configuration on building for your generic module.
I suggest investigating the nodemcu schematic, and the nodemcu specific config parameters for build, in addition to how the three gpio pins must be pulled on bootup.
If you need further assistance, please refer to a community forum, as this is an issue tracker and not meant for general support. See #3655 .