esp8266/Arduino #3863: Re-adding PING after LWIP2 migration?

everslick, 2017-11-22:
As previously discussed ping support was _removed_ by migrating to LWIP2.

see: https://github.com/esp8266/Arduino/pull/3835

Most people might not have noticed yet, as they are still on LWIP1. This issue is meant to gather opinions about re-adding support for it.

Should support for ping be re-added as it was, or would an external ESPing library, that might even work on ESP32, be a better solution?

Or is ping something only very few people care about anyway?

d-a-v, 2017-11-23:
For the record,
 * functions are declared inside espressif SDK
 * it is part of upstream lwIP in lwIP-contrib
 * it can be added in lwip2 (it seems to be a matter of two .c/.h files in a git repo)
 * it will not increase binary size if they are not used

Are you indeed using it ?

everslick, 2017-11-26:
Personally, I'd love to get ping back, by whatever means. I'm curious what the downsides could be, though. I think @igrr was concerned, that it might be harder to support downstream, because it is not in the 'main' source tree anymore, like it used to be for 1.4? but i'm not sure what this means. Is 'lwIP-contrib' not part of the source distribution?

Because I maintain an framework that sits on top of both cores (ESP8266 and ESP32) and abstracts away any differences, I'm especially thankful for APIs that are consistent across platforms. ;-)

d-a-v, 2017-11-26:
lwip-upstream-contrib PING api has changed and is not compatible with declarations in SDK.
So I "forward-ported" lwip-1.4's ping into our lwip2 like it is done for the dhcp server.
You can try it right now with
```
cd tools/sdk/lwip2
make
cd builder
git checkout master
git pull
cd ..
make
```
and please report back whether it is working as before.

bill-orange, 2017-11-26:
I need ping badly.  My ISP furnishes a very poor router in terms of versatility, performance and reliability.  The best way to keep reliably connected that I have found is to PING the router every 6 min. If ping fails, I reset.  Other more common methods like !connected often do not seem to detect that the router is no longer listening.

everslick, 2017-11-26:
@d-a-v: I followed your instructions and ping works the same as it used to do, which is great!

but as soon as i start the mDNS responder, the firmware crashes.

```
0x401043cd: glue2esp_linkoutput at /home/clemens/Devel/ESP/Arduino-ESP8266/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c:263
0x402398dc: etharp_output_LWIP2 at /home/clemens/Devel/ESP/Arduino-ESP8266/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/etharp.c:887
0x4010451c: new_linkoutput at /home/clemens/Devel/ESP/Arduino-ESP8266/tools/sdk/lwip2/builder/glue-lwip/lwip-git.c:206
0x402349bc: ethernet_output at /home/clemens/Devel/ESP/Arduino-ESP8266/tools/sdk/lwip2/builder/lwip2-src/src/netif/ethernet.c:305
0x402398f1: etharp_output_LWIP2 at /home/clemens/Devel/ESP/Arduino-ESP8266/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/etharp.c:893
0x40203d1c: xPortGetFreeHeapSize at /home/clemens/Devel/ESP/emonio/src/../../Arduino-ESP8266/cores/esp8266/heap.c:73
0x4023a450: ip4_output_if_opt_src at /home/clemens/Devel/ESP/Arduino-ESP8266/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c:962
0x402017fc: check_memory at /home/clemens/Devel/ESP/emonio/src/load.c:39
0x40100568: malloc at /home/clemens/Devel/ESP/emonio/src/../../Arduino-ESP8266/cores/esp8266/umm_malloc/umm_malloc.c:1668
0x4023aa84: mem_malloc at /home/clemens/Devel/ESP/Arduino-ESP8266/tools/sdk/lwip2/builder/lwip2-src/src/core/mem.c:136
0x4023a490: ip4_output_if_opt at /home/clemens/Devel/ESP/Arduino-ESP8266/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c:788
0x40239c21: igmp_send at /home/clemens/Devel/ESP/Arduino-ESP8266/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/igmp.c:570
0x40234a18: do_memp_malloc_pool at /home/clemens/Devel/ESP/Arduino-ESP8266/tools/sdk/lwip2/builder/lwip2-src/src/core/memp.c:231
0x40234a54: memp_malloc at /home/clemens/Devel/ESP/Arduino-ESP8266/tools/sdk/lwip2/builder/lwip2-src/src/core/memp.c:231
0x40239cf2: igmp_lookup_group at /home/clemens/Devel/ESP/Arduino-ESP8266/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/igmp.c:570
0x402017fc: check_memory at /home/clemens/Devel/ESP/emonio/src/load.c:39
0x40239ecb: igmp_start_timer at /home/clemens/Devel/ESP/Arduino-ESP8266/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/igmp.c:570
 (inlined by) igmp_joingroup_netif at /home/clemens/Devel/ESP/Arduino-ESP8266/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/igmp.c:521
0x40239f57: igmp_joingroup at /home/clemens/Devel/ESP/Arduino-ESP8266/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/igmp.c:570
0x40222e6a: MDNSResponder::_listen() at /home/clemens/Devel/ESP/emonio/src/../../Arduino-ESP8266/libraries/ESP8266mDNS/ESP8266mDNS.cpp:396
0x40222798: _M_manager at /home/clemens/Devel/ESP/Arduino-ESP8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/functional:1954
0x402018e2: __wrap_free at /home/clemens/Devel/ESP/emonio/src/load.c:82
0x402017fc: check_memory at /home/clemens/Devel/ESP/emonio/src/load.c:39
0x402076dc: operator delete(void*) at /home/clemens/Devel/ESP/emonio/src/../../Arduino-ESP8266/cores/esp8266/abi.cpp:57
0x402241b6: MDNSResponder::begin(char const*) at /home/clemens/Devel/ESP/emonio/src/../../Arduino-ESP8266/libraries/ESP8266mDNS/ESP8266mDNS.cpp:396
0x40222b60: _M_manager at /home/clemens/Devel/ESP/emonio/src/../../Arduino-ESP8266/libraries/ESP8266mDNS/ESP8266mDNS.cpp:396
0x40222f8c: operator() at /home/clemens/Devel/ESP/emonio/src/../../Arduino-ESP8266/libraries/ESP8266mDNS/ESP8266mDNS.cpp:396
 (inlined by) _M_invoke at /home/clemens/Devel/ESP/Arduino-ESP8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/functional:2071
0x4020a963: String::length() const at /home/clemens/Devel/ESP/emonio/src/system.cpp:573
 (inlined by) system_device_name() at /home/clemens/Devel/ESP/emonio/src/system.cpp:389
0x4020d8f6: mdns_start at /home/clemens/Devel/ESP/emonio/src/mdns.cpp:27
 (inlined by) mdns_poll() at /home/clemens/Devel/ESP/emonio/src/mdns.cpp:98
0x4020f3f2: hwpin_poll() at /home/clemens/Devel/ESP/emonio/src/hwpin.cpp:172
0x40207a40: emonio_loop() at /home/clemens/Devel/ESP/emonio/src/emonio.cpp:167
0x40207738: loop at /home/clemens/Devel/ESP/emonio/src/emonio.ino:9
0x40207504: loop_wrapper at /home/clemens/Devel/ESP/emonio/src/../../Arduino-ESP8266/cores/esp8266/core_esp8266_main.cpp:124
0x40100108: cont_norm at /home/clemens/Devel/ESP/emonio/src/../../Arduino-ESP8266/cores/esp8266/cont.S:109
make[1]: Leaving directory '/home/clemens/Devel/ESP/emonio/src'
```
this crash is new.

d-a-v, 2017-11-26:
@everslick thanks for testing ping.
* does the esp8266mDNS/mDNS_Web_Server example works with lwip2  ?
* can you print free heap size just before starting the mDNS responder in your sketch, with lwip1.4 ?
* then with lwip2 ?

devyte, 2018-01-07:
@everslick waiting on feedback.

d-a-v, 2018-01-07:
lwip2 version in core release 2.4.0 has ping.
@bill-orange @everslick can you confirm ping/lwip2 is working for you ?
@everslick please open a new issue if your mDNS bug still occurs with latest git and lwip2.

bill-orange, 2018-01-08:
Sure I can test ping. Just to be sure we are on the same page do you have example code you would like me. To run?

d-a-v, 2018-01-08:
@bill-orange You were saying
> I need ping badly [...] to PING the router every 6 min

so I assumed you were using it in your own sketches and found it no more present in lwip2.

bill-orange, 2018-01-08:
No problem.  I tested it with one of my own sketches.  It works fine with `Ping.ping(WiFi.gatewayIP())`

I am maintaining connectivity with this:

```
if ((millis() - currentLoop) >  pingTest) {
    currentLoop = millis();
    if (Ping.ping(WiFi.gatewayIP())) {
      Serial.println("ping Success!!");
      Serial.print("WiFi - OFF "); WiFiOff(); delay (20); WiFiOn();
      while (WiFi.status() != WL_CONNECTED) {
        delay(250);
        Serial.print(".");
      }
      Serial.println(" WiFi - ON");
    } else {
      Serial.println("ping Error :(");
      ESP.restart();
    }
  }
```
pingTest is set for 15 min.

WiFi On/Off uses this:

```
void WiFiOn() {

  wifi_fpm_do_wakeup();
  wifi_fpm_close();

  //Serial.println("Reconnecting");
  wifi_set_opmode(STATION_MODE);
  wifi_station_connect();
}

void WiFiOff() {

  //Serial.println("diconnecting client and wifi");
  //client.disconnect();
  wifi_station_disconnect();
  wifi_set_opmode(NULL_MODE);
  wifi_set_sleep_type(MODEM_SLEEP_T);
  wifi_fpm_open();
  wifi_fpm_do_sleep(FPM_SLEEP_MAX_TIME);

}

```

closed 2018-01-08 by d-a-v