esp8266/Arduino #3864: Problem with httpUpdate

struthio, 2017-11-22:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			ESP-12E
Core Version:      	?2.1.0-rc2?

### Description

I try to update firmware by httpUpdate OTA after successful flashing (message OK from updater) I reset device but instead of getting running code I get some random characters. Please see Debug Log.

Data in debug:
[begin] _size:             0x0003C7D0 (247760)
MD5 Success: 46a45ed6c4166e05f6a471ca7be06ee6
is correct. This is proper file size and MD5 of file on server.

In PlatformIO I also forced flash size (by build_flags = -Wl,-Tesp8266.flash.4m1m.ld) but this didn't help. After reboot new flash is not booted properly.

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp
#define PROJECT_NAME "Demo"
//#define DEBUG_UPDATER Serial
#define DEBUG_ESP_HTTP_UPDATE 1
#define DEBUG_ESP_UPDATER 1
#define DEBUG_ESP_PORT Serial
#define GPIO_PIN D2

#include <ESP8266WiFi.h>
#include <ESP8266httpUpdate.h>

const char* ssid     = "SSID";
const char* password = "PASS";

static uint8_t iRunUpdate = 0;

void keyPressed()
{
  Serial.println("KeyPressed!");
  iRunUpdate = 1;
}

void updateCheck()
{
  Serial.println("[OTA]: Checking For Updates");
  //"URL", "VERSION=EMPTY", "FINGERPRINT"
  ESPhttpUpdate.rebootOnUpdate(false);
  t_httpUpdate_return ret = ESPhttpUpdate.update("http://domain/" PROJECT_NAME "/firmware.bin");
  switch(ret)
  {
    case HTTP_UPDATE_FAILED:
      Serial.printf("[OTA]: HTTP_UPDATE_FAILD Error (%d): %s", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
      Serial.println();
      break;

    case HTTP_UPDATE_NO_UPDATES:
      Serial.println("[OTA]: HTTP_UPDATE_NO_UPDATES");
      break;

    case HTTP_UPDATE_OK:
      Serial.println("[OTA]: HTTP_UPDATE_OK");
      break;
  }
}

void setup()
{
	Serial.begin(115200);
  // Just delay to be able to connect

  for (int i=5; i > 0; i--)
  {
    Serial.printf("Waiting %i...\r\n",i);
    delay(1000);
  }
  Serial.println(PROJECT_NAME ": " __DATE__ " " __TIME__ );
	Serial.println("------------------");
	Serial.println("Connecting WiFi...");

  WiFi.setOutputPower(50);
  WiFi.setAutoConnect(false);
  WiFi.persistent(false);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print("-");
  }
	Serial.println();

	Serial.print("IP      : "); Serial.println(WiFi.localIP());
	Serial.print("NetMask : "); Serial.println(WiFi.subnetMask());
	Serial.print("RSSI    : "); Serial.println(WiFi.RSSI());

  pinMode(GPIO_PIN, INPUT);
  attachInterrupt(GPIO_PIN, keyPressed, RISING);

  Serial.println("Setup Done!");
  //updateCheck();
}

void loop()
{
  Serial.print(".");
  delay(1000);
  if (iRunUpdate == 1)
  {
    iRunUpdate = 0;
    Serial.println("Updating...!");
    updateCheck();
  }
}

```

### Debug Messages

```
Waiting 5...
Waiting 4...
Waiting 3...
Waiting 2...
Waiting 1...
cESPDemo: Nov 23 2017 00:17:55
------------------
Connecting WiFi...
-----
IP      : XXX.XXX.XXX.XXX
NetMask : 255.255.255.0
RSSI    : -74
Setup Done!
..........KeyPressed!
KeyPressed!
Updating...!
[OTA]: Checking For Updates
[httpUpdate] Header read fin.
[httpUpdate] Server header:
[httpUpdate]  - code: 200
[httpUpdate]  - len: 247760
[httpUpdate]  - MD5: 46a45ed6c4166e05f6a471ca7be06ee6
[httpUpdate] ESP8266 info:
[httpUpdate]  - free Space: 2895872
[httpUpdate]  - current Sketch Size: 249792
[httpUpdate] runUpdate flash...
[begin] roundedSize:       0x0003D000 (249856)
[begin] updateEndAddress:  0x00300000 (3145728)
[begin] currentSketchSize: 0x0003D000 (249856)
[begin] _startAddress:     0x002C3000 (2895872)
[begin] _currentAddress:   0x002C3000 (2895872)
[begin] _size:             0x0003C7D0 (247760)
MD5 Success: 46a45ed6c4166e05f6a471ca7be06ee6
Staged: address:0x002C3000, size:0x0003C7D0
[httpUpdate] Update ok
[OTA]: HTTP_UPDATE_OK
............................................................
<DEVICE-RESET-HERE>
sdl<l<lc|r#c'odo'cpdsdrlxod#g|d#no$l gl o;gcl{o#d;ll`;csdl<l<lc|r#cgodogc8lsd;lxo$#g|d#ngll 'l`o;'cl{o#$;lxl`'sd$|l<lc|;ccgndogc8lrd;l;gbgb8sd$<l<lc|r#cgndogc8lsd;lxod#g|d#no$l ogl o;gcl{o#d;lxl`grd$|l|lc|;ccg'lo
```

teejaydub, 2018-07-02:
Similar symptom to #4876, but could be a different cause.

earlephilhower, 2019-07-25:
Given the age of the bug, the age of the core (2.1!) and a similar closed bug the @teejaydub pointed out, closing.  If this is still happening, please do open up a new issue after trying the latest release (or preferably git head).

closed 2019-07-25 by earlephilhower