esp8266/Arduino #3865: How "Espressif SDK included in this build is under Espressif MIT License" effects the use of Arduino IDE for building Commercial Applications based on ESP8266?

bapidebabrata, 2017-11-23:
I want to build a commercial application for AI ESP8266 Module. According to Espressif MIT License: "Permission is hereby granted for use on ESPRESSIF SYSTEMS ESP8266 only" but ESP8266 core files are licensed under LGPL. I am confused!!! Can anyone please clear this??

https://www.quora.com/unanswered/How-Espressif-SDK-included-in-this-build-is-under-Espressif-MIT-License-effects-the-use-of-Arduino-IDE-for-building-Commercial-Applications-based-on-ESP8266?srid=hsW9h

igrr, 2017-11-23:
*Disclamer: i am not a lawyer and this is not legal advice.*

Espressif SDKs are distributed under Espressif MIT license. This core includes Espressif non-OS SDK. However, it also includes many files which are not part of Espressif SDK, and these files are LGPL licensed.

It might be possible to have a commercial product around LGPL-licensed ESP8266 Arduino core (i know of a few such cases), but you would need to figure out how to comply with the terms of the license — by distributing object files or a static library of your proprietary code, for example.
Alternatively, you can distribute the source code of you sketch/application, if that fits your business model.

closed 2017-11-23 by igrr

bapidebabrata, 2017-11-23:
Thanks.
Are the libraries part of Espressif SDK?
How I would make sure that I am not using part of Espressif SDK?

igrr, 2017-11-23:
> Are the libraries part of Espressif SDK?

I'm not sure which libraries you are referring to, could you please clarify?

> How I would make sure that I am not using part of Espressif SDK?

bapidebabrata, 2017-11-23:
> I'm not sure which libraries you are referring to, could you please clarify?

I am referring to these: https://github.com/esp8266/Arduino/tree/master/libraries

> It's pretty much impossible to build an application for ESP8266 without using at least some parts of Espressif SDK. So you will definitely be using it if you use ESP8266 Arduino core.

Correct me if I am wrong: Nobody is licensed to use ESP8266 Arduino core's examples on AI ESP8266 modules, including the following example: https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WiFi/examples/WiFiWebServer

bapidebabrata, 2017-11-23:
Thanks a lot. You are doing a great job helping others. I appreciate your compassion towards new developers.

Gamadril, 2018-06-24:
> I don't think this is the case. As mentioned in my first reply, i am not a lawyer, but i don't see anything in the license that would prevent you from using these libraries on the ESP8266.

According to https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/src/ESP8266WebServer.h ESP8266WebServer has the GPLv2 license. Means any sketch (commercial or private) must be licensed under GPL as well. That also means anyone using the popular WiFiManager library which is MIT licensed, but needs ESP8266WebServer also has to license his code as GPL. Even EEPROM lib is GPL. IMHO the only way to make a commercial project with Arduino without making it public domain is to rewrite all the popular libraries for own needs...

igrr, 2018-06-24:
WebServer and EEPROM are both LGPL, not GPL.
WiFiManger may very well be MIT, just the final application which integrates WiFiManager and WebServer (plus the ESP8266 Arduino core itself) needs to comply with both licenses.

Gamadril, 2018-06-24:
LGPL and arduino do not fit together. It's just nonsense. Arduino IDE compiles the library code from the source code and links against compiled sketch code. For LGPL each library has to provide compiled code to link against - for each platform, microcontroller type, it's variations and maybe even compiler version... And to be "really" LGPL usable for commercial products it must be ready for dynamic and not static binding

Gamadril, 2018-06-24:
Just to clarify, I'm not the expert and might be wrong, but license problems with GPL/LGPL just make my head explode 😀

devyte, 2018-06-24:
I'm also not a lawyer or expert in licences, but I have been reading up on it.
We don't use Arduino code, or even Arduino libs. We implement our own core, which happens to be (mostly) compatible with the Arduino api. So whatever licenses apply for Arduino do not apply for us.
In addition, the SDK is licensed under an MIT-like license, which is very permissive, as long as the sdk-specific code isnused only on the ESP8266 system.
Finally, this core is licensed under LGPL, which does allow commercial applications, and is in fact MIT compatible. LGPL does not force you to surrender the user's app source code. The sole restriction is that that app must be distributed in binary form, so that if a user wants to modify this core, and rebuild the final firmware, he can do so ([source](https://www.gnu.org/licenses/gpl-faq.en.html#LGPLStaticVsDynamic)).
It is GPL which seems to make no sense, because if one piece is licensed under GPL, it forces the entire final firmware to be GPL, which means the entire source code must be made available, and the sdk source is closed. Example: [ESP8266 new pwm](https://github.com/StefanBruens/ESP8266_new_pwm) repo. In such cases, distribution is not allowed, as it would violate GPL, so such source code is ok only for private use.