esp8266/Arduino #3867: ESP.deepsleep losing time

YaddyVirus, 2017-11-23:
### Basic Infos

#### Hardware
Hardware:		NodeMCU 1.0
Core Version:      	?2.1.0-rc2?

### Description

What you see below is the screenshot of my NodeMCU writing temperature and light intensity to a MySQL database and going to deep sleep in between. I've programmed it to read the sensors and send the data to the table and goto deep sleep, then repeat the cycle every 60 minutes.

As you can see in the LastUpdate column, each update is 2-3 minutes earlier than the previous one. Why is this happening and how do I ensure that the readings are every 60 minutes only because as you can see the updates are about 41 minutes early already.

I've written my deep sleep command like this - ESP.deepSleep(60000000*60);

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP12E Module)
Flash Size: 		4MB(3M SPIFFS)
CPU Frequency:	80Mhz
Flash Frequency:	40Mhz
Upload Using:		SERIAL

### Sketch

```cpp

#include <ESP8266HTTPClient.h>
#include <ESP8266WiFi.h>
#include <Adafruit_MQTT.h>
#include <Adafruit_MQTT_Client.h>
#include <DHT.h>

/* DHT AND LDR SETUP */

#define DHTTYPE DHT11
#define DHTPIN  0

const int LDR = A0;
int input_val = 0;

/* WIFI SETUP */

#define WLAN_SSID       "ssid"
#define WLAN_PASS       "Password"

/* ADAFRUIT IO SETUP */

#define AIO_SERVER      "io.adafruit.com"
#define AIO_SERVERPORT  1883                   //8883 for SSL
#define AIO_USERNAME    "YaddyVirus"      
#define AIO_KEY         "AIO_KEY"

DHT dht(DHTPIN, DHTTYPE); 
 
float temp_f;  // Values read from sensor
String webString="";     // String to display
unsigned long previousMillis = 0;        // will store last temp was read
const long interval = 2300;              // interval at which to read sensor

/************ Global State******************/

// Create an ESP8266 WiFiClient class to connect to the MQTT server.
WiFiClient client;
// or... use WiFiFlientSecure for SSL
//WiFiClientSecure client;

// Store the MQTT server, username, and password in flash memory.

const char MQTT_SERVER[]     = AIO_SERVER;
const char MQTT_USERNAME[]   = AIO_USERNAME;
const char MQTT_PASSWORD[]   = AIO_KEY;
Adafruit_MQTT_Client mqtt(&client, MQTT_SERVER, AIO_SERVERPORT, MQTT_USERNAME, MQTT_PASSWORD);

/************************* Feeds ***************************************/
const char TEMP_FEED[] = AIO_USERNAME "/feeds/data";
const char LIGHT_FEED[] = AIO_USERNAME "/feeds/light";
Adafruit_MQTT_Publish temp = Adafruit_MQTT_Publish(&mqtt, TEMP_FEED);
Adafruit_MQTT_Publish light = Adafruit_MQTT_Publish(&mqtt, LIGHT_FEED);

/********************** Sketch Code ************************************/
void MQTT_connect();

void setup() {
  Serial.begin(115200);
  delay(10);

  dht.begin();           // initialize temperature sensor

  Serial.println(F("MQTT Temp"));

  // Connect to WiFi access point.
  Serial.println(); Serial.println();
  Serial.print("Connecting to ");
  Serial.println(WLAN_SSID);

  WiFi.begin(WLAN_SSID, WLAN_PASS);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println();

  Serial.println("WiFi connected");
  Serial.println("IP address: "); Serial.println(WiFi.localIP());

  temp_f = dht.readTemperature();
  input_val = analogRead(LDR);
  Serial.println();
  Serial.print("Initial Temp: ");
  Serial.println(temp_f);
  Serial.println();
  Serial.print("Initial Light:");
  Serial.println(input_val);
  Serial.println();

  // Loop begins

    // Ensure the connection to the MQTT server is alive (this will make the first
  // connection and automatically reconnect when disconnected).
  MQTT_connect();

  HTTPClient http;
  http.begin("http://203.185.191.35/test3.php");
  http.addHeader("Content-Type", "application/x-www-form-urlencoded");  //Specify content-type header
  int httpCode = http.POST("Sensor_ID=MBoxTest_1&temp="+String(temp_f)+"&light="+String(input_val));
  String payload = http.getString();
  Serial.println(httpCode);
  Serial.println(payload);
  http.end();
                             
  // Publishing now...
    temp_f = dht.readTemperature();
    input_val = analogRead(LDR);
    Serial.print(F("\nSending temp: "));
    Serial.print(temp_f);
    Serial.print("...");
    
    Serial.print(F("\nSending Light: "));
    Serial.print(input_val);
    Serial.print("...");
    if (! temp.publish(temp_f) || ! light.publish(input_val)) {
      Serial.println(F("Failed"));
    } else {
      Serial.println(F("Sent!"));
    }
  
  // ping the server to keep the mqtt connection alive
  // NOT required if publishing once every KEEPALIVE seconds

  if(! mqtt.ping()) {
    mqtt.disconnect();
  }

  ESP.deepSleep(60000000*60);
}

void loop() {

}

// Function to connect and reconnect as necessary to the MQTT server.
// Should be called in the loop function and it will take care of connecting.
void MQTT_connect() {
  int8_t ret;

  // Stop if already connected.
  if (mqtt.connected()) {
    return;
  }

  Serial.print("Connecting to MQTT... ");

  uint8_t retries = 3;
  while ((ret = mqtt.connect()) != 0) { // connect will return 0 for connected
       Serial.println(mqtt.connectErrorString(ret));
       Serial.println("Retrying MQTT connection in 5 seconds...");
       mqtt.disconnect();
       delay(5000);
       retries--;
       if (retries == 0) {
         // basically die
         while (1);
       }
  }
  Serial.println("MQTT Connected!");
}
```

### Debug Messages

```
messages here
```

[image]

caverna, 2017-11-23:
``` // Loop begins ``` inside ```setup()``` function !?!?

YaddyVirus, 2017-11-23:
@caverna that's a stupid little comment I wrote for my own reference... Actually that comment marks the beginning of the code that was initially written in the loop function before I implemented deep sleep

5chufti, 2017-11-29:
print the millis() before you ESP.deepSleep().
You allways have the time to connect WiFi, connect to server etc. and then start the 60min deepsleep.
Allthough that might not add up to the 2 min, it will ba quite a lot. And from experience the internal timer is off for ~5%.

RudyFiero, 2017-12-05:
> As you can see in the LastUpdate column, each update is 2-3 minutes earlier than the previous one. Why is this happening and how do I ensure that the readings are every 60 minutes

http://www.espressif.com/sites/default/files/2c-esp8266_non_os_sdk_api_guide_en_v1.5.4.pdf
22. system_rtc_clock_cali_proc 
RTC clock period tends to drift with change in temperature, so RTC timer is not very precise. 

The RTC in the ESP8266 does not use a crystal for timing. An oscillator that is based on a capacitor is not going to be very accurate. Sure there can be compensation but there can be a number of factors that can cause drift and deviation. There are no guarantees as far as accuracy from the RTC. It is what it is. This is a cheap part that does so much. But it does not do everything perfectly. If you need precise timing then include a external Real Time Clock in your designs.

5chufti, 2017-12-05:
depending on your energy budget if only ~sec accurate timing is relevant you can wake up early, fetch a sntp timestamp and use that for more exact timing. e.g. for "on every 5min"
```
  time_t = time(nullptr);
  int k = 300 - (t % 300);
  delay(k * 1000);
  t += k;
```

YaddyVirus, 2017-12-05:
@RudyFiero Sounds like adding an external RTC might be the only option.

Also @5chufti I tried doing what you suggested earlier. For some reason it produces random deviation in time between readings

d-a-v, 2019-10-01:
There's nothing we can do.
For this kind of problem one should have a server for tagging the measurement at the time of receiving.
Or the ESP can be resynchronized with SNTP on wake up before sending.
Or an external RTC if ESP is waking up to measure & store before going to sleep again, several times before sending everything to a server.

closed 2019-10-01 by d-a-v