esp8266/Arduino #3869: network connect fails while serial IF recieves burst

s0170071, 2017-11-23:
Please fill the info fields, it helps to get you faster support ;)

### Basic Infos
I have a MT681 power meter with an IR output connected to the UART. The meter sends bursts of 500 bytes every 1200ms @9600bps. So one burst lasts about 500ms.

#### Hardware
Hardware:			WemosD1 mini
Core Version:      	2.4.0rc2

### Description
When the RX bursts are present upon boot, the ESP fails to connect to wifi. It does connect eventually but it takes it a unpredictable amount of time, mostly about 10min. 
Workaround: 
Set UART mode to tx only, RXbuffersize =0. Wait for connect , then adjust the buffer size and the uart mode according to your needs.

### Settings in IDE

Module:  			Wemos D1mini
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		SERIAL (tried OTA too)
Reset Method:		power cycle

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

s0170071, 2017-12-22:
may be related to #2576

s0170071, 2018-03-19:
I don't mean to rush anyone but could anyone please have a look ? At least try the sketch and confirm this is a bug.

d-a-v, 2018-03-20:
You may try #4536.

earlephilhower, 2019-01-16:
Closing because #4328 is merged, and no feedback for many months.

closed 2019-01-16 by earlephilhower