esp8266/Arduino #3871: ESP crash when large webpages are accessed via vpn

s0170071, 2017-11-23:
### Basic Infos

#### Hardware
Hardware:			wemos D1 mini
Core Version:      	2.4.0rc2

### Description

If I serve a 8kb webpage from my ESP to a computer in my lan or my mobile phone it works fine. 
If I activate vpn on the phone, the website does not load. Sometimes it hangs a while, sometimes it crashes if you retry.
If I limit the packet size to 1460 and wait 20ms after the packets it works. See workaround sample code.

When using the sketch, access <ip> and it crashes when accessed via vpn. Open <ip>/workaround and you get a webpage.
My LAN setup: 

internet <-> router1 <-> router2 <-> esp
                                           ^
android phone ----VPN ----I
The vpn is tunneled through router1.

Router1 has router2 in its DMZ and has mtu limited to 1400.
Router2 (AVMFritzbox7490) has no mtu setting. They say its automatic.

### Settings in IDE

Module:  			WemosD1mini
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		SERIAL
Reset Method:		power cycle

### Sketch

```cpp
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>

const char* ssid = "SSID";
const char* password = "*****";

ESP8266WebServer server(80);

void setup(void){
  Serial.begin(115200);
  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  if (MDNS.begin("esp8266")) {
    Serial.println("MDNS responder started");
  }

  server.on("/", crashme);
  server.on("/workaround", workaround);

  server.begin();
  server.setContentLength(CONTENT_LENGTH_UNKNOWN);
  Serial.println("HTTP server started");
}

void loop(void){
  server.handleClient();
}
void crashme()
{
  String webpage = ""; 
  for (int i = 0; i<800;i++) webpage +="0123456789";
   server.send(200, "text/plain", webpage);
  }
  void workaround()
{
  server.setContentLength(CONTENT_LENGTH_UNKNOWN);
  server.sendHeader("Cache-Control","no-cache");
  server.sendHeader("Transfer-Encoding","chunked");
  server.send(200,"text/plain");
  String webpage1 = ""; 
  String webpage2 = ""; 
  String webpage3 = ""; 
  String webpage4 = ""; 
  String webpage5 = ""; 
  for (int i = 0; i<140;i++){ // string needs to be <1460bytes
   webpage1 +="0123456789";
   webpage2 +="0123456789";
   webpage3 +="0123456789";
   webpage4 +="0123456789";
   webpage5 +="0123456789";
  }
 
   server.sendContent( webpage1); delay(20); // delay >= 20ms
   server.sendContent( webpage2); delay(20);
   server.sendContent( webpage3); delay(20);
   server.sendContent( webpage4); delay(20);
   server.sendContent( webpage5); delay(20);
   server.sendContent("");
  }
```

### Debug Messages

```
messages here
```

devyte, 2017-11-24:
@s0170071 please try latest git with lwip2.

s0170071, 2017-11-24:
sorry, tried for several hours but just cant get the latest git version to work. 
I use the arduino ide and did what was written here: https://github.com/esp8266/Arduino#using-git-version.
Still when opening the boards manager, it lets me only choose those core versions that were there all along.
Strange thing though: the 2.4.0rc2 package on my machine is in C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2 rather than the the arduino program/hardware folder. Of course I have tried to drop the current files there too, even tried to exchange the 2.4.0rc2 for the current ones. That was giving a error when compiling: unknown version. 
Tried atom with platformio too, didnt work either. Cant even choose a core version at all. Did compile but still I fail to see the beauty of platformio :-O. Does linter a bit but can't jump to code when clicking a compile error  :-(

Can you provide a arduino IDE boards manager url pointing to the latest git ?

s0170071, 2017-11-24:
compiled with Atom and platformio. Used these settings in platformio.ini:
`
[env:d1_mini]
;platform = espressif8266
board = d1_mini
framework = arduino
platform = https://github.com/platformio/platform-espressif8266.git#feature/stage
`
Not sure if it used the most current version. I find that platformio configuration and output is very obscure. Anyway, the result is the same. Does not work.
Each time I call the IP address (-> crashme) via vpn I lose 13k of ram and the function blocks for about 2 secs.

devyte, 2017-11-25:
Pio probably builds against lwip1.4, unless you do something special. I don't know enough about pio to guide you. From the Arduino IDE it's as easy as selecting an option in the menu.

s0170071, 2017-11-25:
> From the Arduino IDE it's as easy as selecting an option in the menu.

then by all means tell me. I've been trying all day. It is probably just a matter of entering the correct boards manager url. I use
`https://github.com/esp8266/Arduino/releases/download/2.4.0-rc2/package_esp8266com_index.json`
and that one only lets me select the releases.

devyte, 2017-11-25:
Latest git doesn't install via board manager. Instructions are on readthedocs.

s0170071, 2017-11-25:
That I figured. Can you be a bit more specific, please? I tried for hours and  tend to give up on this.

Pablo2048, 2017-11-25:
OMG another lazy one - he means this https://github.com/esp8266/Arduino#using-git-version :-(

s0170071, 2017-11-25:
Thank you very much. Thats exactly the motivation I was looking for. Besides that: the manual doesnt work for Windows7.  I think I am done here, got a family to attend to...

Pablo2048, 2017-11-25:
I'm on W7 and installed github version without problem so what is your problem installing git version of the core on W7 EXACTLY?

s0170071, 2017-11-25:
It is not showing up in the board manager after I installed it.

Pablo2048, 2017-11-25:
What is "Not showing up"? Github version will not show up in Board manager list of installed cores (obviously because its not installed via JSON file). Does every step from the manual installation finish correctly? Did you try to install Arduino IDE in portable variant and try to install into this one to prevent overwriting your current configuration?

s0170071, 2017-11-25:
Yes, it installed  correctly- at least without errors and is sitting in its directory. Odd thing though, the 2.3.0 version is installed in 
C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0
whereas the IDE is in 
C:\Program Files (x86)\Arduino.
I tried to put the git version in 
C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266
and
C:\Users\John\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0
Neither worked. Or at least I couldnt tell?

If the git version is not showing in the board manager, how can I select it / switch it on / can tell that it is active ?
Thank you.

Pablo2048, 2017-11-25:
IMHO there cannot be another version installed together with the github one. Please do as I suggest - install Arduino IDE in portable version (after unpacking .zip file create folder named "portable" and run Arduino.exe - if you need more information STFW ( https://www.arduino.cc/en/Guide/PortableIDE )) and try to install Github version into this one. Do not install any other Core via Board manager. Then give a try to your code...
BTW: your example code is really not friendly to stack/heap :-(
To do this more stack/heap friendly what about for example this:
String webpage = "";
webpage.reserve(8100); 
for (int i = 0; i<800;i++)
  webpage.concat(F("0123456789"));

s0170071, 2017-11-25:
Finally. Portable did the trick. And the problem is gone. No crash.
Still, having latest git as an option in the board manager would be so much easier.  
Thank you for your support.

closed 2017-11-25 by s0170071