esp8266/Arduino #3881: ArduinoOTA interferes with hardware serial

djdehaan, 2017-11-26:
Hi all,

Not a question, but a tip for others who struggle. I use a Wemos D1 with a GPS receiver and SD-card. And I did not get the ArduinoOTA function to work. IT started uploading but after 10 or 12 or 8 bullets appearing in the IDE it stopped uploading. After long searching I found the reason. 

Earlier, I found it most stable if I address that GPS using hardware serial, meaning I connect it on the RX/TX pins. Appearantly, that interferes with the upload process. When I add the following code in the Setup() routine, the upload works like a charm:

```
  ArduinoOTA.onStart([]() {
    Serial.end();
  });
  ArduinoOTA.setHostname("CarTrax");
  ArduinoOTA.begin();
```
Maybe this is completely logic, but it took me some time to find out and I did not find similar cases online.

lrmoreno007, 2017-11-29:
WOW I try it and now it work like a charm. Before it failed 50% of the time or more.

Now this is my setupOTA function:
```
void setupOTA() {
  ArduinoOTA.setHostname(hostOTA);

  //ArduinoOTA.setPort(8266);

  // No authentication by default
  //ArduinoOTA.setPassword((const char *)"123");

  ArduinoOTA.onStart([]() {
    Serial.println("Comienzo OTA");
    Serial.end();
  });
  ArduinoOTA.onEnd([]() {
    Serial.begin(115200);
    Serial.println("\nOTA Finalizado con exito");
  });
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    //Serial.printf("Subido: %u%%\r", (progress / (total / 100)));
  });
  ArduinoOTA.onError(
      [](ota_error_t error) {
        Serial.begin(115200);
        Serial.printf("Error[%u]: ", error);
        if (error == OTA_AUTH_ERROR) Serial.println("Fallo de autentificacion");
        else if (error == OTA_BEGIN_ERROR) Serial.println("Fallo al continuar");
        else if (error == OTA_CONNECT_ERROR) Serial.println("Fallo de conexion");
        else if (error == OTA_RECEIVE_ERROR) Serial.println("Fallo de recepcion");
        else if (error == OTA_END_ERROR) Serial.println("Fallo final");
      });
  ArduinoOTA.begin();
}
```

igrr, 2017-12-27:
Possible duplicate of #2576 #3869.

d-a-v, 2018-01-01:
Could you please try again with latest git and without onStart's Serial.end() ?

lrmoreno007, 2018-01-04:
For me with latest git today and hand fixing #4086 (index problem and int<-> char warning), it work fine without Serial.end(). Tested many times successfully.

Regards

devyte, 2018-01-04:
Ok, relevant PRs are merged. Please retest.

lrmoreno007, 2018-01-05:
Working OK for me, uploading sketch and SPIFFS image.

Only sometimes it remains as "Uploading" and after a while the message "An error occurred while sending the sketch" appears, but I remember that this was a problem of the Arduino IDE. However, the sketch has been sent correctly if we look at the serial output.

Tested with Arduino IDE 1.8.3 and 1.8.5.

devyte, 2018-01-05:
@djdehaan will you be testing this as well?

djdehaan, 2018-01-05:
@devyte I will, I hope to find time this weekend.

djdehaan, 2018-01-06:
@devyte I have a test-setup, however, I do not know how to download your upgraded lib. I did in GitGui a Remote / FetchFrom / Origin, that does update something but not ArduinoOTA. When running a test (with serial.end removed) it goes wrong again. To be expected with the old libs. How do I get your new libs?

devyte, 2018-01-06:
@djdehaan I'm not sure about gitgui, I didn't even know it existed. I use git from (linux) command line.
The repo has to be clones with git into a specific directory for the arduino build system to pick it up. Also, the core installed via board manager has to be uninstalled, both can't coexist.
Instructions for git install are on readthedocs.

djdehaan, 2018-01-06:
@devyte Indeed, ESP8266 is not installed via board manager. But following the (Windows) instructions:

git clone https://github.com/esp8266/Arduino.git esp8266
cd esp8266/tools
python get.py
Restart Arduino

So I deleted the entire directory and reinstalled ESP8266. It will not do OTA without   Serial.end();
[image]

d-a-v, 2018-01-10:
Does this bug happen too with the example Files/Examples/ArduinoOTA/BasicOTA,
or only with your sketch ?

earlephilhower, 2019-01-31:
There is no working MCVE, and there were some big changes to the uart infrastructure and OTA (including re-enabling interrupts while flash is being written which allows serial comms even during the long flash ops you'd see in OTA, and uart logic corrected to not write out of bounds in overflow conditions) in 2.5.0.  Please try the latest 2.5.0 release and open a new issue w/MCVE if there is still an issue.

closed 2019-01-31 by earlephilhower