esp8266/Arduino #3882: UDP packets not sending after waiting for > 5 seconds

999chris, 2017-11-27:
Hardware:			ESP-12E
Core Version:      	Unkown

Sending UDP packets is okay-ish if I send a packet every 2-5 seconds. But I don't need to send stuff that frequently, say 30 seconds... With a 30 second delay (loop is still running, with yields) the first packet sends (upon boot) but then nothing else is received by Wireshark/Node.js app.

I have two ESP12(E's I think). One of which is using almost identical code (albeit an older version and this broadcasts to 255.255.255.255 and really has no problems. This one however does not. 192.168.0.255 (or similar) only.)

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	40Mhz
Upload Using:		OTA / SERIAL via Arduino ATMega
Reset Method:		ck

Sketch (in short, i've removed uninteresting functions etc):

```
#include <dummy.h>
#include <SPI.h>
#include <U8g2lib.h>
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>

byte secondDelay = 30;
long timeOut = 0;
long count = 0;

WiFiUDP Udp;
IPAddress broadcast(192, 168, 0, 255);
const int UDP_PORT = 3141;
char incomingPacket[255];  // buffer for incoming packets

void setup() {

WiFi.begin(XXXX,XXXX);

  Serial.print("Connecting");
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
  Serial.println();

  Serial.print("Connected, IP address: ");
  Serial.println(WiFi.localIP());

  deviceID = WiFi.localIP()[3];

  if (Udp.begin(UDP_PORT)) {
    Serial.println("UDP Port established");
  };

//SOME MORE INIT STUFF, THEN YIELD.

  yield();

}

void loop() {

  long timeNow = millis();

  reading = float(analogRead(readingPin)) / 1024; //Convert to Voltage Float 0.00 - 1.00
  reading = ((reading * 2 * 1000) - 500) / 10; //Curve on specs of TGZ
  temp_float += reading;
  count++;

  if (timeNow >= timeOut) {

    timeOut = timeNow + (secondDelay * 1000);
 
    temp_float /= count;
    
    if (Udp.beginPacket(broadcast, UDP_PORT)) {

      Serial.println("Writing Packet");

      dtostrf(temp_float, 5, 2, tempStr);

      Udp.write(tempStr, sizeof(tempStr));

      Udp.endPacket();
      yield();
    }

    writeToScreen(int(temp_float));

    count = 0;
    temp_float = 0;

  }

  int packetSize = Udp.parsePacket();

  if (packetSize)
  {
    // receive incoming UDP packets
    Serial.printf("Received %d bytes from %s, port %d\n", packetSize, Udp.remoteIP().toString().c_str(), Udp.remotePort());
    int len = Udp.read(incomingPacket, 255);
    if (len > 0)
    {
      incomingPacket[len] = 0;
    }
    Serial.printf("UDP packet contents: %s\n", incomingPacket);
    yield();
    //writeToScreen(String(incomingPacket[0]).toInt());
  }

  yield();

}
```

devyte, 2017-11-27:
@999chris please specify core version and lwip version used.

999chris, 2017-11-27:
Hi.

Thanks for replying. How does one find this information. Iwip?

Information in this email including any attachments may be privileged, confidential and is intended exclusively for the addressee. The views expressed may not be official policy, but the personal views of the originator. If you have received it in error, please notify the sender by return e-mail and delete it from your system. You should not reproduce, distribute, store, retransmit, use or disclose its contents to anyone. Please note we reserve the right to monitor all e-mail communication through our internal and external networks. SKY and the SKY marks are trademarks of Sky plc and Sky International AG and are used under licence.

Sky UK Limited (Registration No. 2906991), Sky-In-Home Service Limited (Registration No. 2067075) and Sky Subscribers Services Limited (Registration No. 2340150) are direct or indirect subsidiaries of Sky plc (Registration No. 2247735). All of the companies mentioned in this paragraph are incorporated in England and Wales and share the same registered office at Grant Way, Isleworth, Middlesex TW7 5QD.

d-a-v, 2017-11-27:
If you are using master, it's in tools/lwip-variant menu.
If you are not using master, you are using lwip-1.4.

999chris, 2017-12-01:
Couple of delays dotted around sorted this.

closed 2017-12-01 by 999chris