esp8266/Arduino #3884: If WiFi.softAP() after WiFi connected, then Blynk can't work.

bestpika, 2017-11-27:
### Basic Infos

#### Hardware

Hardware: `WeMos D1 mini`
Core Version: 7b09ae5f69f8dccb5152d128c8484ec9a18adb11

### Description

If WiFi.softAP() after WiFi connected, then Blynk can't work.
But 5c01841430d5f93d5334c56f3b708995912a8675 is okay.

### Sketch

Not work:
```cpp
#define BLYNK_PRINT Serial

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <BlynkSimpleEsp8266.h>

void setup()
{
  Serial.begin(115200);
  Serial.println();
  WiFi.mode(WIFI_AP_STA);
  WiFi.begin("STA ssid", "STA passphrase");

  int wait = 0;
  while (!WiFi.isConnected() && wait++ < 10) {
    Serial.print(".");
    delay(1000);
  }
  WiFi.softAP("AP ssid", "AP passphrase");

  Blynk.config("Blynk Token", "blynk-cloud.com");
  Blynk.connect();
}

void loop()
{
  Blynk.run();
}
```

Work:
```cpp
#define BLYNK_PRINT Serial

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <BlynkSimpleEsp8266.h>

void setup()
{
  Serial.begin(115200);
  Serial.println();
  WiFi.mode(WIFI_AP_STA);
  WiFi.begin("STA ssid", "STA passphrase");

  WiFi.softAP("AP ssid", "AP passphrase");
  int wait = 0;
  while (!WiFi.isConnected() && wait++ < 10) {
    Serial.print(".");
    delay(1000);
  }

  Blynk.config("Blynk Token", "blynk-cloud.com");
  Blynk.connect();
}

void loop()
{
  Blynk.run();
}
```

d-a-v, 2017-11-27:
Can you retry with core version 7b09ae5 (latest master) and with lwip-v1.4 instead of v2 in menus ?

bestpika, 2017-11-27:
@d-a-v I can't find lwip-v1.4
[image]

bestpika, 2017-11-27:
I try `Prebuilt Source (gcc)`, then it work.
[image]

```txt
wifi evt: 2
wifi evt: 0
wifi evt: 3

[9869] 
    ___  __          __
   / _ )/ /_ _____  / /__
  / _  / / // / _ \/  '_/
 /____/_/\_, /_//_/_/\_\
        /___/ v0.4.10 on Arduino

[9871] Connecting to blynk-cloud.com:8442
[hostByName] request IP for: blynk-cloud.com
[hostByName] Host: blynk-cloud.com IP: 188.166.206.43
[10050] Ready (ping: 75ms).
wifi evt: 7
wifi evt: 7
wifi evt: 7
```

d-a-v, 2017-11-27:
> [image]

There is v2, and the others which all are v1.4.
Try the first one (Prebuilt), it was the default before v2 appeared.

d-a-v, 2017-11-27:
So lwip-v2 is the cause. I'll work on it. Thanks for the report !

d-a-v, 2017-11-27:
I can't reproduce it. It works with both sketch.
Can you remove your ```++wait<10``` test and re-check ?

d-a-v, 2017-11-27:
Apparently my softAP password was not long enough :]
I finally could reproduce the bug. I also added the simple web server in your sketch.
Oddly, with lwip2 + webserver and softAP after STA, blinky does not work but webserver works on both interfaces.

tcpdump shows the ARP request and answer for blynk-cloud.com resolving.

Now digging into Blynk lib...

d-a-v, 2017-11-27:
@bestpika Are you trying with exactly the sketches above ?
I could reproduce the bug only because I forgot to remove an ```ESP.eraseConfig()``` lost in my tryings.

Both your sketches are OK with lwip2 here.
I need more input, or someone else try.

It may be not related but I also noticed blink services takes 5 seconds more everytime I reset the ESP right after beeing connected (I guess it is a fair protection against buggy device).

bestpika, 2017-11-27:
I will try tomorrow.
By the way, I using Arduino IDE 1.8.5 and Windows 10 1703 x64.

devyte, 2017-11-27:
My quick test seems to agree with @d-a-v . 
Closing as can't reproduce.

closed 2017-11-27 by devyte

bestpika, 2017-11-28:
Today I try over 20 times.
Not work still not work.

d-a-v, 2017-11-28:
Are you using exactly the sketch above with your IDs ?

bestpika, 2017-11-28:
@d-a-v yes.

d-a-v, 2017-11-28:
@devyte I think this is an interesting issue with lwip-v1.4 working where lwip2 does not, so I think it is worth investigating a bit more.
@bestpika can you activate lwip2 log and retry ?
you need to:
```
cd tools/sdk/lwip2
make
```
edit ```builder/glue/gluedebug.h``` and change ```ULWIPDEBUG``` from 0 to 1, then
```
make install
```
In your sketch, add ```Serial.setDebugOutput(true);``` right after ```Serial.begin()```, then restart your failing sketch.
If you can post the console somewhere so we can check the IP traffic.