esp8266/Arduino #3887: Nodemcu 0.9 Wdt reset with Keypad.h library on getKey()

ynohtna92, 2017-11-28:
### Basic Infos

#### Hardware
Hardware:		ESP-12
Core Version:      	?

### Description

The program does not complete running the first loop and instantly initiates a wdt reset.
It seems to have issues with the getKey() function in the <Keypad.h> arduino library.

### Settings in IDE

Module:  			Nodemcu v0.9
Flash Size: 		4MB
CPU Frequency:       80Mhz
Flash Mode:		?
Flash Frequency:	?
Upload Using:		Serial 115200
Reset Method:		nodemcu

### Sketch

```cpp
#include <Keypad.h>

const byte ROWS = 4; //four rows
const byte COLS = 3; //three columns
char keys[ROWS][COLS] = {
  {'1','2','3'},
  {'4','5','6'},
  {'7','8','9'},
  {'*','0','#'}
};
byte rowPins[ROWS] = {1,2,3,4}; //connect to the row pinouts of the keypad
byte colPins[COLS] = {5,6,7}; //connect to the column pinouts of the keypad
Keypad keypad = Keypad( makeKeymap(keys), rowPins, colPins, ROWS, COLS );

void setup(){
  Serial.begin(115200);

  pinMode(LED_BUILTIN, OUTPUT);

  Serial.println("MagicLottoPrinter Client");
  Serial.println("------------------------");
}
  
void loop(){
  digitalWrite(LED_BUILTIN, HIGH);
  delay(250);
  Serial.println("Loop");
  digitalWrite(LED_BUILTIN, LOW);
  delay(250);
  yield();
  char key = keypad.getKey();
  if (key != NO_KEY){
    Serial.print("Key pressed: ");
    Serial.println(key);
  }
}
```

### Debug Messages

```
 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
```

devyte, 2017-11-28:
@ynohtna92 you are using gpios 1-7 to connect to the keypad. Some of those are forbidden, e.g.: serial port, or flash.
Please study the pinout for the nodemcu board. Keep in mind that GPIOi != Dj, i.e.: indices for GPIO are different than for D. There are #defines for Dj pin nomenclature, which map to gpio indices. Even then, be careful with what you use, e.g.: don't use D3.
Closing due to usage error.

closed 2017-11-28 by devyte

ynohtna92, 2017-11-29:
Thanks, I followed the pinout mapping in this post https://github.com/esp8266/Arduino/issues/584

This is what @devyte  is referring to for anyone viewing this.

```
#define D0 16
#define D1 5 // I2C Bus SCL (clock)
#define D2 4 // I2C Bus SDA (data)
#define D3 0
#define D4 2 // Same as "LED_BUILTIN", but inverted logic
#define D5 14 // SPI Bus SCK (clock)
#define D6 12 // SPI Bus MISO 
#define D7 13 // SPI Bus MOSI
#define D8 15 // SPI Bus SS (CS)
#define D9 3 // RX0 (Serial console)
#define D10 1 // TX0 (Serial console)
```